realsenseai/librealsense #10769: D455: Suggestions for use with repetitive surfaces

DanMesh, 2022-08-12:
| Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu  | 

### Recommendations/suggestions for observing repetitive surfaces?

In certain situations, I would like to use the RealSense D455 for perceiving fairly uniformly textured surfaces, such as stony or grassy ground. However, the resulting depth images are often very sparse or "patchy" (see example below).

Are there best practices for improving the depth perception in these scenarios, or other suggestions for how to increase the amount of depth points per image?

I am using the configuration parameters suggested in [this white paper](https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones).

All ideas are appreciated!

---

#### Example images
[image]
_**Left:** infrared image, **Right:** depth image_
_The depth image is very sparse and "patchy"_

MartyG-RealSense, 2022-08-12:
Hi @DanMesh  Very grainy, rough textured surfaces like the ones in your image would usually be very good for obtaining depth information from.  The sparseness of the depth image and the reference to the drone paper makes me think that this image may be taken from an elevated position some meters above the ground.  Is that correct, please?

If it is correct and you have the camera's IR Emitter enabled then it is possible that the sparse depth image is resulting from the semi-random dot pattern that the camera projects not being able to reach the ground.  The camera uses the dots that are cast onto surfaces to analyze them for depth information.

You could try slightly extending the range of the dot pattern projection by maximizing the **laser power** option to '360'.  But the higher power of an external IR dot pattern projector may be needed for longer-range projection, which would likely be too heavy and impractical for a flying drone with no mains electricity power source.

The camera can alternatively use any visible or near-visible light in a scene for analyzing surfaces for depth information instead of the dots, but the apparent downward orientation of the camera may be limiting the amount of light that reaches it.

In the JSON configuration listing in the drone paper, the parameter **controls-laserpower** is set to '0', meaning that the projector is disabled and no dot pattern is being projected.  This typically results in the kind of depth image that you have unless ambient light in the scene can make up for the absence of the dots.

DanMesh, 2022-08-15:
Thanks for the swift reply @MartyG-RealSense.
You're right: these images were taken from ~10m above the ground with a downward-facing camera.
The stereo receivers have ND filters (18% transmission) attached to _reduce_ the amount of incoming light and prevent overexposure in bright sunlight.

In my setup, the IR transmitter is off as the projected pattern is usually not visible in bright light and at large distances. As you say, adding an external projector isn't feasible, so I am interested in seeing how I might improve the depth image with my current setup.

Are there certain parameters (in the `.json` or elsewhere) that would be worth adjusting to try and improve this image?

MartyG-RealSense, 2022-08-15:
If over-exposure is a concern then if the ND filters are removable, you could instead try setting an auto-exposure Region of Interest (ROI) in the lower half of the image to help to avoid light saturation of the sensors.  https://github.com/IntelRealSense/librealsense/issues/2190#issuecomment-589027974 provides information about defining a ROI as a rosparam in the ROS1 launch file.

MartyG-RealSense, 2022-08-15:
**param-secondpeakdelta** is set to 775 in the drone paper json settings.  This is likely to make the SDK very selective about which depth values it considers accurate enough to allow to be displayed on the image.  The default value of this setting is only 325.  So you could try 325 to see how much of the depth detail is added.

DanMesh, 2022-08-17:
Thanks for that suggestion @MartyG-RealSense. I'll keep investigating this and post an update on what I find.

MartyG-RealSense, 2022-08-17:
Thanks very much, @DanMesh - I look forward to your next report.  Good luck!

MartyG-RealSense, 2022-08-29:
Hi @DanMesh  Do you have an update about this case to provide, please?  Thanks!

DanMesh, 2022-08-30:
@MartyG-RealSense No, not yet. I still need to try adjusting `param-secondpeakdelta`, so there has been no change at this point.

MartyG-RealSense, 2022-08-30:
Okay, thanks very much for the update.

DanMesh, 2022-09-08:
@MartyG-RealSense To follow up on this issue, I have two things to report:
1. Decreasing `param-secondpeakdelta` from 775 to 325, as you suggested, made a noticeable difference to the image quality and greatly improved the fill level of the depth images.
2. Swapping the camera for a different device (same model) made an even bigger difference. I suspect the camera used to capture the images in the initial description was becoming miscalibrated, hence the patchy images.

From my side, the issue seems to be solved, and I will go ahead and recalibrate the old camera using the guide here: https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide

closed 2022-09-08 by DanMesh

MartyG-RealSense, 2022-09-08:
Thanks so much @DanMesh for the detailed update about what worked for you - it's great to hear that you achieved a solution!