realsenseai/librealsense #10770: Convert raw depth data to rs.object

pixml27, 2022-08-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | .... | 
| Operating System & Version |   Linux (Ubuntu 20.04)   | 
| Kernel Version (Linux Only)    |  (5.4.8)                                         | 
| Platform                                 | PC |
| SDK Version                            |  ...  | 
| Language                            |  {python }                          | 
| Segment			|  { others }                          | 

### Issue Description

Hi guys!
We are trying to repeat [this](https://dev.intelrealsense.com/docs/box-measurement-and-multi-camera-calibration) example
However, we don't have access to live cameras, all we have is stored infrared images, depth images and raw depth data.

Our only problem is we can't apply [filters ](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L53) to the depth image because we have raw data and not an rs2::frame object.

So, is there a way to convert the numpy array loaded from the xxx_Depth.raw file into the rs.stream.depth object that is being used [here](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/python/examples/box_dimensioner_multicam/calibration_kabsch.py#L156)?

[image]

MartyG-RealSense, 2022-08-12:
Hi @pixml27  If your goal is to convert a numpy array to rs2::frame format in Python, this has previously been explored by RealSense users but remains an unsolved problem unfortunately as far as I am aware, as described in https://github.com/IntelRealSense/librealsense/issues/2551

Whilst the RealSense SDK's **software device** interface can be used, it works best with C++ and has been shown to have difficulties with using it in Python scripts.

pixml27, 2022-08-13:
@MartyG-RealSense thank you for you answer!
Ok, I will not use post-processing - I have questions further
What are the values in raw depth data measured in, cause i'm sure it is not meters!
[image]

I need to use two functions: [first](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py#L99) and [second](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py#L121)
So i want to implement `depth_frame.as_depth_frame().get_distance` by myself - do you have any advices how can i do it?)

MartyG-RealSense, 2022-08-13:
Raw 16-bit 'pixel depth' values (**uint16_t**) like those in the above image can be multiplied by the **depth unit scale** of the camera to convert them to the real-world distance in meters. For all 400 Series models except the D405 the default depth unit scale is **0.001** and does not changed unless altered deliberately by the program or by user input.

For example, for a raw depth value of 3735, multiplying it by 0.001 gives a real-world distance of **3.375 meters**. 

If the get_distance() instruction is used though then you do not have to multiply by the depth scale as the instruction does it for you and provides the result in meters.

https://github.com/IntelRealSense/librealsense/issues/9939 should be a helpful reference for Python scripting for get_distance and for information about the difference between uint16t * depth scale and obtaining depth in meters with get_distance()

pixml27, 2022-08-14:
@MartyG-RealSense Thanks for the tips and links - they helped a lot!

closed 2022-08-26 by MartyG-RealSense