realsenseai/librealsense #10777: D435i: How to generate PCL Point Cloud at faster rate?

KainanSu, 2022-08-15:
Like the issue that's been closed. #7676

I want to process pointcloud with pcl, but the example in this repo convert data format to pcl costs lots time. (600ms)
In #7676, that is say [rs-gl](https://github.com/IntelRealSense/librealsense/tree/master/examples/gl) may use gpu help to accelerate  converting.

But I meet 2 problem as follow:
1. I cannot find anything help to convert with gpu, but just use gpu to store pointcloud(`rs2::gl::pointcloud`). I dont think it can help me to accelerate.
2. [Latest SDK](https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0) doesnt include `librealsense2-gl/rs_processing_gl.hpp`.

Could you give any solutions? Thanks.

MartyG-RealSense, 2022-08-15:
Hi @KainanSu  Pointclouds can be a processing intensive operation, so reducing the burden on the CPU (lowering its 'percentage usage) by offloading the processing work onto a GPU can in itself improve an application's performance.

GLSL can work with **RS2::** instructions simply by inserting **GL::** into the instruction.  For example for pointclouds, rs2::pointcloud becomes **rs2::gl::pointcloud**

An example of rs2::pointcloud in a RealSense PCL pointcloud script is **rs-pcl-color**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L158

However, if the computer is a low-power device that does not have a strong GPU then using GLSL may not provide a noticeable performance boost.

More information about GLSL is at https://github.com/IntelRealSense/librealsense/pull/3654

The file that you are searching for is in the **librealsense > include >librealsense2_gl** folder of the SDK in the latest **master** version.

https://github.com/IntelRealSense/librealsense/tree/master/include/librealsense2-gl

KainanSu, 2022-08-15:
@MartyG-RealSense Hi, Marty! Thanks for such a quick response!

I have reinstall the SDK twice, but I cannot see `librealsense2_gl` in `C:\Program Files (x86)\Intel RealSense SDK 2.0\include`. There is just `librealsense2` In the foder.

When I install SDK, I had select full install. Buy the way , I use win10, and CUDA 9.0.

MartyG-RealSense, 2022-08-15:
If you installed the SDK on Windows with the Assets file list's **Intel.RealSense.SDK-WIN10** installer file, there is another link in the list called **Source code - zip**.  If that zip file is downloaded and extracted then there should be a similar folder and file structure to the GitHub website version of the SDK that I linked to.

The SDK can be built from source code on Windows using instructions at the link below, but as far as I am aware GLSL support should be in the SDK version installed by the Intel.RealSense.SDK-WIN10 installer file too.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

CUDA support requires a computer with an Nvidia GPU, whilst GLSL is 'vendor neutral' and should work with any GPU brand.  

If you have a Windows PC with an Nvidia video card then I would think that CUDA support could be enabled by building the SDK from source code with CMake using the instructions link above and setting the CMake build flag **BUILD_WITH_CUDA** to true.  I'm not certain though, as virtually all librealsense cases that use CUDA are on computers running Linux, mostly on Nvidia Jetson computing boards that have a built-in Nvidia GPU.

KainanSu, 2022-08-15:
@MartyG-RealSense Thanks, Marty. I try it right now.

KainanSu, 2022-08-16:
@MartyG-RealSense Hi, Marty! I looked into the [convert code](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/pcl/pcl-color/rs-pcl-color.cpp) and I found the function [RGB_Texture](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L66) costs most time(80%).
I try to rewrite `RGB_Texture` and now it can capture frame in 200ms, it is already enough for me.

MartyG-RealSense, 2022-08-16:
That's great to hear that you achieved a fast-working solution, @KainanSu - thanks very much for the update!

MartyG-RealSense, 2022-08-21:
Case closed due to solution achieved and no furtther comments received.

closed 2022-08-21 by MartyG-RealSense

FeiSong123, 2023-02-18:
Hello, I heard that you have rewritten RGB_Texture function then improves the speed of point cloud acquisition. Can you share your rewritten function, please？ @KainanSu

KainanSu, 2023-02-19:
@FeiSong123 Sorry, I cannot find the code. But I remember I the code I rewrite were below: reduce function calls, reduce object unnecessary recreations.

FeiSong123, 2023-02-19:
All right. Thank you for your help. I'll go back and try your way.

FeiSong123, 2023-02-19:
Hello, I tried the method you said, and then successfully reduced the time of point cloud acquisition. Thank you for your help! @KainanSu