realsenseai/librealsense #10781: Software genlock trigger?

antithing, 2022-08-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)                                    | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Is it possible to trigger the frame capture with a software command?

I have a third party hardware sync pulse that I read in code. I need to trigger the realsense shutter when this pulse is received in c++.

Thanks!

MartyG-RealSense, 2022-08-16:
Hi @antithing  A simple method of accomplishing your goal may be to use trigger the C++ script provided at https://github.com/IntelRealSense/librealsense/issues/2219 when your C++ code acknowledges the third-party trigger pulse. This script takes a single image capture from all attached cameras simultaneously.

The script has a mechanism that only allows it to activate if a D415 camera model is detected, so that checking mechanism would need to be edited out if your RealSense model is not a D415.

*******

Alternatively, you could use the **External Synchronization (Genlock)** hardware sync system described in the Intel white-paper guide at the link below.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

The Genlock system allows a camera to be configured as a **Genlock slave** by setting its Inter Cam Sync Mode option to a value of '4'.  The camera will then wait indefinitely for a trigger signal from an external signal generator device and only initiate capture when it detects that pulse.

The status of Inter Cam Sync Mode can be set with scripting.  For example in C++:

**sensor.set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 4);**

antithing, 2022-08-16:
Thank you! i am trying to avoid any hardware development (the genlock signal comes over an SDI cable currently, into a capture card) so the first option sounds good. 

i am a bit confused by the snippets in the linked thread. 

So i would do this? (psuedo)

```
void run()
{

//connect camera
while(true)
{
          if(syncpulse== true)
          {
                                  frameset fs;
			          if (kvp.second.poll_for_frames(&fs))
			          {
				          //process frames
			          }
          }
}
}
```

i need to stream the imu data as well, in between camera frame grabs, is there any example code available on that setup?

Thanks again

MartyG-RealSense, 2022-08-17:
Your psudeo-code is structured logically, yes, so it should work.

The **rs-data-collect** C++ example program can collect IMU data (accel and gyro) if the attached camera has an IMU.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

MartyG-RealSense, 2022-08-23:
Hi @antithing  Do you require further assistance with this case, please?  Thanks!

antithing, 2022-08-23:
All good! thank you very much.

closed 2022-08-23 by antithing

MartyG-RealSense, 2022-08-24:
Thanks very much @antithing for the update!

tugbakara, 2022-09-01:
> The Genlock system allows a camera to be configured as a **Genlock slave** by setting its Inter Cam Sync Mode option to a value of '4'. The camera will then wait indefinitely for a trigger signal from an external signal generator device and only initiate capture when it detects that pulse.

Hi @MartyG-RealSense , we want to create map from rtabmap_ros and we have two D435i, and I our map is nor quite well, we have a doubt about hw sync, do we have to do that in order to create map? If it is, our first consideration is to sync depth and RGB that's why, I thought that we can make FULL SLAVE mode, but in documentation only GENCLOCK sync type is documented and I don't really get it FULL SLAVE, do we still have to trigger from external signal generator as in GENCLOCK or only setting params in rs-viewer or in SDK is enough?

MartyG-RealSense, 2022-09-01:
Hi @tugbakara  Hardware sync enables the timestamps of multiple cameras to follow the timing of a central trigger.  It is not necessary to use hardware sync for all multiple camera applications though.  An example of this is Intel's ROS guides for stitching together pointclouds from two or more RealSense cameras at the links below.

**Two cameras, one computer**
https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

**Three cameras, two computers**
https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

***********

Full Slave mode is enabled when Inter Cam sync Mode is set to '3', and this mode synchronizes RGB and depth streams **on the same camera** (i.e not synching to an external non-RealSense RGB camera).

As you highlight though, there is only a small amount of documentation for Full Slave mode.  It is probable that you would be able to achieve your map generation goal without using hardware sync.

tugbakara, 2022-09-01:
> **Two cameras, one computer**
> https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

In that section, we are not aiming to unite two pointclouds only, we are trying to use RGB and depthto make [map](https://www.youtube.com/watch?v=JYYE82xNa8I) (as an example), I reallly don't get why did you point out that?

MartyG-RealSense, 2022-09-01:
You said above that "we want to create map from rtabmap_ros and we have two D435i".  I interpreted this as you wanting to combine together the pointcloud views of two D435i cameras into a single combined pointcloud map.  I apologize if I misunderstood.

Intel's single-camera D435i SLAM guide that uses rtabmap_ros would produce a similar map to the one in the YouTube video that you linked to.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

tugbakara, 2022-09-01:
Thanks, we will look at that tutorial and if it works great and if we get some trips and tricks , I will provide here :)