realsenseai/librealsense #10783: Unity: how to ignore errors in depth stream due to defective cam

jeffrson, 2022-08-16:
I'm using an L515 cam with latest firmware (1.5.8.1) in Unity on Windows 10.

The cam appears to be defective (the depth sensor, afaict). Unfortunately I need to continue with development using this cam until the replacement will arrive.

My questions regard some behaviour I found and if I could work around it for some time.

In "RealSense Viewer", when I enable "RGB Camera", after a couple of seconds, I get a black frame with "No Frames Received!". OTOH, when I first enable "L500 Depth Sensor" (which shows two stream windows for depth and infrared, also blinking "No Frames Received" on it), and only then also enable "RGB Camera" the camera picture is shown alongside the erroneous Depth Sensor streams.

In Unity 2021.3.8f1, with .unitypackage of SDK 2.50 (although I tried previous versions of both to no avail) I get a message
```
ExternalException: rs2_pipeline_wait_for_frames(pipe:0000029454FC5010)
Rethrow as Exception: Frame didn't arrive within 5000
```

So, out of curiosity, why does "RealSense Viewer" depend on the order of sensor activation? More importantly however, is there any way to get a similar behaviour in Unity: use RGB camera picture and ignore error of Depth sensor for the time being?

MartyG-RealSense, 2022-08-17:
Hi @jeffrson  Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end). 
 Does doing so make any difference?

MartyG-RealSense, 2022-08-24:
Hi @jeffrson  Do you require further assistance with this case, please?  Thanks!

jeffrson, 2022-08-24:
Thanks @MartyG-RealSense - I had tried reversing the connector's orientation before. I didn't help then and now.

I wasn't able to to read the RGB picture with the defective camera to help with in-house development. I tried to work around with some screenshots, which is far from effective, but I had no choice.

I finished the project for now, so I don't need further assistance at the moment, although the question remains unanswered. Feel free to close though.

Thanks!

MartyG-RealSense, 2022-08-24:
Thanks very much @jeffrson for the update!  As you suggested, I will close the case.  Thanks again!

closed 2022-08-24 by MartyG-RealSense