realsenseai/librealsense #10785: No Frames Recieved in INTEL Real Sense D435i.

vgunashekhar, 2022-08-16:
[image]
[image]
[image]
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|047322070697|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|047423051767|
|**Firmware Update Id**|047423051767|

I tried to install complete sdk files also
Please provide a possible solution to the problem

MartyG-RealSense, 2022-08-17:
Hi @vgunashekhar  Does the problem with **No Frames Received** still occur if you reboot the computer and then open the RealSense Viewer and only enable depth and RGB (and not enable the Motion Module)

If the problem still occurs if the Motion Module is not enabled, could you provide information about the following questions please.

1.  Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice?  The official cable should still work in a USB 2.1 port, though operate at USB 2 speed (hence the '2.1' beside the camera name in the side-panel).

2.  Did you install the RealSense SDK using the automated installer file **Intel.RealSense.SDK-WIN10** in the 'Assets' file list on the SDK **Releases** page?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

3.  Has the camera been stored unused for a period of weeks / months before the problem was found to occur?

MartyG-RealSense, 2022-08-24:
Hi @vgunashekhar  Do you require further assistance with this case, please?  Thanks!

vgunashekhar, 2022-08-24:
>    1. Has the camera been stored unused for a period of weeks / months

Thanks for the response
I am using a generic C -type cable. But it works good in my all other
laptops. Both in Linux and windows 10.
I downloaded the package you mentioned in your email's image also but it's
not working.
I want to ask one more question that is it related to the processor also.
Because my other laptops in which I tested were i5 and the one in which it
is not working is i7.
Thanks and regards
V.Gunashekhar

MartyG-RealSense, 2022-08-24:
Is the i7 laptop older than the i5 ones?  With processors, their model code can be more important than the 'i' number because the first number of the model code describes how old the processor is.

For example, an  i7‑6785R has the first number '6' in its 6785R code, meaning it is a '6th generation' processor that dates from around 2016.  An i7‑11850HE has '11' as its first number, meaning that it is an '11th generation' processor that dates from 2021 and so has significantly superior technology.

vgunashekhar, 2022-08-26:
No, it's not. My i7 is 11th generation. It's model number is i7-11370H.
My other i5 laptops are older generation like 6th and 7th.
Thanks & regards
V.Gunashekhar

MartyG-RealSense, 2022-08-26:
Pllease check next whether the SDK's **rs-capture** example program is able to run on Windows on your 11th generation laptop by launching it using the instructions at https://github.com/IntelRealSense/librealsense/issues/7544#issuecomment-707577574

vgunashekhar, 2022-09-01:
Yes, I tried but still didn't work. So, I switched to another Camera for
the time being because I have to complete my project soon. I will get back
again on this issue.
Thanks for the support
Regards
V.Gunashekhar

MartyG-RealSense, 2022-09-01:
You are very welcome.  I look forward to your next report.  Good luck!

closed 2022-09-23 by MartyG-RealSense