realsenseai/librealsense #10786: RTAB Map - D435i Resolution

tugbakara, 2022-08-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.13.0.50 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    | 4.9.253| 
| Platform                                 |  NVIDIA Jetson AGX Xavier  |
| SDK Version                            | 2.50  | 
| Language                            |                            | 
| Segment			|  Robot   | 

Hi, in the rtab map package of the ros , we used this package with d435i and we 
mapped our office (**note**: before the proccess we made custom calibration). I provide the map of the office when d435i is moving with our robot but there is doubt about the mapping is OK or do we have to change the params of [robot_localization package](https://github.com/cra-ros-pkg/robot_localization) , or do we have to make calibration again? The reason of doubts actually are coming from some areas of the map, here are they:
[image]

[image]

[image]

[image]

And in the last there is a table but everytime when we try to make map of table it always corrupted and we are stuck, could you give an idea about the problems that we might missed. (I provide the actual photo of it).

map of table:
[image]

photo of table : 
[image]

Thanks in advanced! :)

MartyG-RealSense, 2022-08-17:
Hi @tugbakara   The apparent glossy surface of the floor raises the possibility that the floor section of the images is being distorted by light reflecting off the floor.  Once the floor distortion is resolved then we can determine whether it is causing problems in the other areas of the image.

The negative effects of glare from reflections can be significantly reduced by adding a physical optical filter product called a **linear polarization filter** over the lenses on the outside of the camera.  This is discussed in Section **4.4  When to use polarizers and waveplates** of Intel's white-paper document about optical filters.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#4-the-use-of-optical-filters

Any thin-film polarizer filter that is linear should work (except for the circular ones in 3D glasses) and so they can be obtained inexpensively by searching stores such as Amazon for the term **linear polarizer sheet**

The discussion at the link below is also a useful reference about reducing the effects of reflectivity from a glossy floor.

https://support.intelrealsense.com/hc/en-us/community/posts/360043612734-Issue-with-tiled-floor

tugbakara, 2022-08-17:
Thanks @MartyG-RealSense , I will look what you attached here tomorrow. I will provide feedback ASAP. Have a nice day! 🙂

MartyG-RealSense, 2022-08-24:
Hi @tugbakara  Do you have an update about this case that you can provide, please?  Thanks!

tugbakara, 2022-08-24:
Hi @MartyG-RealSense ! We suspend this task temporarily . We are planning to turn back one or two week later, we didn't buy polarized sheet yet. When we make something worthy,I will provide the outcomes and if the issue is not issue anymore, I will close it. I am online here, I won't forget to close it . 
Have a nice day :)

MartyG-RealSense, 2022-08-24:
Thanks, @tugbakara  :)

tugbakara, 2022-09-12:
I am closing the issue, because we couldn't pass that level yet. When we try it, i will update. Have a nice day!

closed 2022-09-12 by tugbakara

MartyG-RealSense, 2022-09-12:
Thanks very much @tugbakara   :)