realsenseai/librealsense #10788: Unhandled exception escaping from a worker thread!

vstadnytskyi, 2022-08-16:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2022-08-17:
Hi @vstadnytskyi  Which RealSense camera model do you have, please?

Does the error still occur if you follow the procedure below?

1.  Install the older RealSense Viewer version **2.48.0** by visiting the link below and downloading and running the file **Intel.RealSense.Viewer.exe** in the 'Assets' file list?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

[image]

2.  After launching the 2.48.0 Viewer, close it and launch the 2.50.0 version of the Viewer again.

MartyG-RealSense, 2022-08-24:
Hi @vstadnytskyi  Do you require further assistance with this case, please?  Thanks!

vstadnytskyi, 2022-08-24:
Thank you, @MartyG-RealSense  for a speedy reply. The problem was with the faulty RAM on the windows 11 PC. It caused many strange behaviors including issues with Intel RealSense viewer.

After running memtest86 on the computer and extracting faulty memory, the issue with Intel RealSense disappeared.

Thank you!

MartyG-RealSense, 2022-08-24:
It's great to hear that you achieved a solution, @vstadnytskyi - thanks very much for the update!

Lumos2050, 2023-07-04:
@MartyG-RealSense Thank you!

MartyG-RealSense, 2023-07-04:
@Lumos2050  You are very welcome  :)