realsenseai/librealsense #10789: I want to get Y16 frame on ROS2 Driver.

demul, 2022-08-17:
Camera Model : D455
OS : Ubuntu 18.04
ROS Version : eloquent

@MartyG-RealSense 
I can't use the intel custom calibration tool.
I must get Y16 stream inside of my automatic calibration system.
So i must get Y16 stream on API level(ros2).
Please, Would u kindly try to get Y16 infra image on ros2 driver on D455 and tell me how to do it?
Please try it and teach me how to reproduce. (Docker is ok)
Thank u

demul, 2022-08-17:
https://github.com/IntelRealSense/realsense-ros/issues/1670
This link do not works for me because it's ros1 driver,
And, Editting "base_realsense_node.cpp" directly do not works for me.
The stream type never change!(It's always Y8 rectified image)

MartyG-RealSense, 2022-08-17:
Closed as a duplicate of https://github.com/IntelRealSense/librealsense/issues/10766

Hi @demul  Let's continue to discuss the issue at the above case link.  Thanks!

closed 2022-08-17 by MartyG-RealSense