realsenseai/librealsense #10791: Get vertices from differents frames

Amayess, 2022-08-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0 | 
| Language                            | C | 
| Segment			|  Robot | 

### Issue Description

Hi @MartyG-RealSense,

I want to acquire vertices from different frames, so I made a loop that roughly looks like that:

```
for (int i = 0; i < NbofFramesWanted ; i++)
{
    // Get frames
    auto frames = pipe.wait_for_frames();
    // Generate the pointcloud and texture mappings
    points = pc.calculate(depth);    // Récupération des vertices    auto vertices = points.get_vertices();

    /* store vertices with another loop using vertices[i].x vertices[j].y and vertices[j].y */ 
}
```

Before that loop I start the pipeline streaming with pipe.start() and after the loop I end the stream with pipe.stop(). I noticed that all the vertices of all the frames I get are the same. Which is odd because I thought wait_for_frames() was supposed to get me new frames.

So I tried putting the pipe.start() and pipe.stop() inside my loop and now I get different actual different vertices between all the frames. However this slows down the execution time by a lot and I feel like this is not the proper solution.

Do you know why I get these results ?

Thank you.

MartyG-RealSense, 2022-08-17:
Hi @Amayess  May I first ask which camera firmware driver version you are using, please?  In firmware version **5.13.0.50** there was a fix for an issue where sequential depth start-stops could cause intermittent failures.

[image]

Amayess, 2022-08-18:
Hi @MartyG-RealSense, the camera firmware driver version is : 5.13.00.50

MartyG-RealSense, 2022-08-18:
If you are starting and stopping on each loop instead of letting the script run without stopping then I wonder whether the **for (int i = 0; i < NbofFramesWanted ; i++)** condition highlighted in blue below could be removed. 

[image]

Jackarnd, 2022-08-18:
Hello @MartyG-RealSense I'm part of @Amayess team.
We are running the script only once. What we want is:

1. Grab n frames,
2. Get their vertices,
3. Store the vertices of each frame in `x;y;z` arrays. 
4. do stuff with that

That's a straight forward approach hence the for loop. We noticed that the stream didn't "reset" or "update" if we didn't start and stop it after each iteration (which takes a while)... I thought that `frames = pipe.wait_for_frames();` was supposed to update `frames` with new acquired frames? We haven't tried poll_for_frames as the behavior is expected to be the same, but maybe that's the solution...?
What is the expected behavior here?

MartyG-RealSense, 2022-08-19:
The version of the documentation for the **rs-pointcloud** SDK C++ example program at the link below has some additional information that suggests code for printing the pointcloud.

https://dev.intelrealsense.com/docs/rs-pointcloud

```
/* this segment actually prints the pointcloud */
auto vertices = points.get_vertices();              // get vertices
auto tex_coords = points.get_texture_coordinates(); // and texture coordinates
for (int i = 0; i < points.size(); i++)
{
    if (vertices[i].z)
    {
        // upload the point and texture coordinates only for points we have depth data for
        glVertex3fv(vertices[i]);
        glTexCoord2fv(tex_coords[i]);
    }
}
```

A key difference between this code and yours is that after getting the vertices, this script uses **points.size();** in the for-int instruction instead of 'NbofFramesWanted'

Jackarnd, 2022-08-19:
Well we do use `points.size()` to actually get our vertices data (I'm sorry I didn't add it earlier). Below you can find our loop to store the vertices data which is located where @Amayess wrote `/* store vertices with another loop using vertices[i].x vertices[j].y and vertices[j].y */`  in the original ticket.
```
auto vertices = points.get_vertices(); // Vertices declaration
Size = points.size(); // Size of the vertices array

// Loop to store vertices data
for (int j = 0; j < Size ; j++) 
{
        // Storing vertices data
        Cloud[j].x = vertices[j].x;
        Cloud[j].y = vertices[j].y;
        Cloud[j].z = vertices[j].z;
 }
```

> The expected behaviour of wait_for_frames() and poll_for_frames is described at [#2422 (comment)](https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709)

I have seen this comment which led us to create this ticket. Because I don't understand the behavior we have in our code: why we have to "reset" the pipe stream (with `start()` and `stop()` each time we want a new frame) in order to get new frames while `wait_for_frame()` is supposed to update with new frames...

I'm guessing I'm missing something here but I'm not sure what.

Now that I'm thinking about, the key difference between our code and the usual code we see in the documentation is in the loop, usually it's a while loop that runs up until the `app` is over so it goes in many other "main" realsense functions but for us it's a simple loop. Maybe that's the reason?

MartyG-RealSense, 2022-08-19:
If the number of times that for-int is allowed to loop is defined by **NbofFramesWanted**, have you defined a value for NbofFramesWanted ?  Otherwise its default value would be '0'.

Here is another approach for setting up a for loop and wait_for_frames:

**for (auto i = 0; i < 30; ++i) pipe.wait_for_frames();**

This comes from Intel's C++ **save-to-disk** example, which captures only the first 30 frames and then saves the last frame as an image and stops.

https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp

MartyG-RealSense, 2022-08-25:
Hi @Amayess and @Jackarnd  Do you require further assistance with this case, please?  Thanks!

Jackarnd, 2022-08-25:
We figured the first 30 frames or so are pretty much identical (which may be linked to auto-exposure) which caused our problem. We still have to check if the frames after that are really different though. But it seems that the problem came from there.

MartyG-RealSense, 2022-08-25:
Thanks very much @Jackarnd for the update.  I look forward to hearing about your next test results.  Good luck!

MartyG-RealSense, 2022-09-06:
Hi @Jackarnd  Do you have an update about this case that you can provide, please?  Thanks!

Jackarnd, 2022-09-06:
@MartyG-RealSense The original problem from this post seems to be solved. However we discovered some other problems but we will create another ticket for that later. Thank you.

MartyG-RealSense, 2022-09-06:
Thanks very much @Jackarnd for the update!  As the problem described in this case appears to be solved, I will close it as you suggested.  Please do create another ticket later for your other questions.  Thanks again!

closed 2022-09-06 by MartyG-RealSense