realsenseai/librealsense #10792: Apt server is broken/down, gives 403 Forbidden

devany, 2022-08-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | Intel® RealSense™ SDK 2.0) | 
| Operating System & Version |   Ubuntu 22.04.1 LTS  | 
| Kernel Version (Linux Only)    | Linux 5.15.0-46-lowlatency x86_64                                    | 
| Platform                                 |  |
| SDK Version                            |  2.0                          | 
| Language                            |  n/a | 
| Segment			|  n/a | 

### Issue Description
Following the instructions at:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

I receive:
Err:6 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
  403  Forbidden [IP: 2600:9000:218c:c000:1a:eb13:66c0:93a1 443]

I notice there are similar issues that are now closed #9349 - but the solutions there do not work for me.

Thanks & regards

MartyG-RealSense, 2022-08-18:
Hi @devany  On a 22.04 Jammy installation discussion at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 there is an unofficial test patch to add support for kernel 5.13 and 5.15, and also a pre-built DKMS test package.

devany, 2022-08-18:
But I can't download the SDK, so I can't apply a patch.

Also, or in any case, the instructions I originally followed have now broken apt such that I always receive a forbidden IP error when attempting apt update... item 2 on the instructions.

MartyG-RealSense, 2022-08-18:
Have you expanded open the 'Assets' file list on that instruction page that contains the file quoted in item 2, please?

[image]

devany, 2022-08-18:
Yes, I have that file, but i can't successfully **apt update** without throwing "Forbidden IP".

i.e.
`sudo apt-get update
Hit:1 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Get:3 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [99,8 kB]
Hit:4 https://dl.google.com/linux/chrome/deb stable InRelease
Get:5 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]                             
Err:6 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
  403  Forbidden [IP: 2600:9000:218c:5400:1a:eb13:66c0:93a1 443]
Reading package lists... Done                             
E: Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/jammy/InRelease  403  Forbidden [IP: 2600:9000:218c:5400:1a:eb13:66c0:93a1 443]
E: The repository 'https://librealsense.intel.com/Debian/apt-repo jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
`

MartyG-RealSense, 2022-08-18:
I will discuss your problem with my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2022-08-19:
Hi @devany  My colleagues provided the following information about your issue.

*********

Ubuntu 22.04.1 LTS is not yet formally supported in LRS, so the apt repository not available yet.  The customer can install the pre-release Debian packages with dpkg commands

```
sudo dpkg -i <deb package to be installed>
sudo dpkg -r <name of the deb package to be removed>
```

There are two ways to resolve the Forbidden IP error:

*  Method #1 - remove the realsense apt server from apt with command below:

`sudo add-apt-repository --remove "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`

Then sudo apt update should work.

*  Method #2 - remove the realsense apt repository server from apt manually:

Modify the `/etc/apt/sources.list` file directly, delete the lines with the realsense apt repository, save the file back.

Check the repo list with this command:

`cat /etc/apt/sources.list`

MartyG-RealSense, 2022-08-25:
Hi @devany  Do you require further assistance with this case, please?  Thanks!

devany, 2022-08-25:
Thanks for all your help, the issue can be closed now.

Cheers!

closed 2022-08-25 by devany