realsenseai/librealsense #10796: Cannot get IMU data into python 3.7.10 on Windows 10 laptop

r-zonis, 2022-08-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version | Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.50.0.3785                          | 
| Language                            |  python                        | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I recently purchased a D435i RGBD camera and am trying to stream depth, RGB, and IMU data from the camera. When I try to do that, I get the following TimeOutError:

"Frame didn't arrive within 5000"
 
Note that I tried doubling the timeout period and this did not do anything. When I comment out the added IMU streaming channels and stream only depth and RGB data, then everything works fine. If I do the opposite i.e stream only IMU data, I get the same timeout error.

From my review of the logs of issues that have been raised over the past few years, I see that this has been an on again off again issue for a couple years now. The fact that this issue still seems to persist is truly astounding, considering one of the primary functions of the D435i IS TO PROVIDE IMU DATA!! There is nothing more frustrating than paying good money for a product and not just having it work right out of the box, then needing to spend HOURS fumbling around the internet looking for a solution.

Code can be found here:
`import os
import numpy as np                        # Fundamental package for scientific computing
import pyrealsense2 as rs
import matplotlib.pyplot as plt
import time

class RealsenseCamera:
    
    def __init__(self, data_types_of_interest=['depth', 'RGB', 'IMU']):
        self.data_types_of_interest = data_types_of_interest
        print("Searching Camera..")
        try:
            self.camera = rs.context().devices[0]
            print('Camera type:', self.camera.get_info(rs.camera_info.name))
        except IndexError:
            raise IndexError("No RealSense device detected!")
        print("Available sensors for", self.camera.get_info(rs.camera_info.name))
        for s in self.camera.sensors:  # Show available sensors in each device
            print(s.get_info(rs.camera_info.name))
        self.acquiring = False

    #Currently (08/18) I am having trouble retrieving IMU data from the camera. IMU functionality has been validated using Intel
    #RealSense viewer, which is testing program creating by Intel. I posted an issue on the GitHub page for librealsense, the code
    #base for interacting with the camera (pyrealsense2 is a part of librealsense), so we will see what comes of it.
    def begin_data_acquisition(self):
        if self.acquiring == True:
            print('Data acqusition has already begun!')
            return
        self.acquiring = True
        print("Beginning data acqusition")
        self.pipe = rs.pipeline()
        self.cfg = rs.config()
        self.cfg.disable_all_streams()
        #Done so that we are only acquiring the data types we are interested in. Makes data acquisition faster
        if 'depth' in self.data_types_of_interest:
            self.cfg.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
        if 'RGB' in self.data_types_of_interest:
            self.cfg.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
        if 'IMU' in self.data_types_of_interest:
            self.cfg.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 200)  # acceleration
            self.cfg.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)  # gyroscope
        print('Skipping first few frames so camera can calibrate...')
        self.profile = self.pipe.start(self.cfg)
        for _ in range(10):  # Skip first frames to give syncer and auto-exposure time to adjust
            self.pipe.wait_for_frames(timeout_ms=10000)
        # # while 1:
        for _ in range(10):
            self.frame_set = self.pipe.wait_for_frames()
            if 'depth' in self.data_types_of_interest:
                self.depth_frame = self.frame_set.get_depth_frame()  # Get depth frame
            if 'RGB' in self.data_types_of_interest:
                self.RGB_frame = self.frame_set.get_color_frame()
            if 'IMU' in self.data_types_of_interest:
                self.accel_frame = self.frame_set.first_or_default(rs.stream.accel, rs.format.motion_xyz32f)
                self.gyro_frame = self.frame_set.first_or_default(rs.stream.gyro, rs.format.motion_xyz32f)`

MartyG-RealSense, 2022-08-18:
Hi @r-zonis  The Python IMU discussion at https://github.com/IntelRealSense/librealsense/issues/8492 will be a good place to start if you have not seen it already.  The RealSense user in that case was also using Windows 10 and Python 3.7.  A range of IMU script references are provided at that case which initially did not work for the user, and they also experienced the **Frame didn't arrive within 5000** error.  

They found that all of the IMU scripts in the discussion worked on another PC and so the problem was likely with some aspect of the configuration of the PC that they were initially testing with.

r-zonis, 2022-08-18:
Hi @MartyG-RealSense thanks for the reference. So is the answer that there is no answer? Are there debug logs somewhere that might give me info on more precisely why the system is stalling? I am only testing this code on a Windows laptop; ultimately, in it's final application, image acquisition will occur through a Linux PC. I guess for now, unless there something I can do to better track down this problem, I will just ignore and hope that it works on Linux.

MartyG-RealSense, 2022-08-18:
I would recommend trying the scripts in that discussion on Windows to see whether the same error also occurs in those scripts too.

closed 2022-08-31 by MartyG-RealSense