realsenseai/librealsense #10799: librealsense-2.40.0/src/media/ros/ros_reader.cpp:514:Wpessimizing-move

dcb314, 2022-08-21:
Recent development versions of gcc can be made to say:

/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/media/ros/ros_reader.cpp:514:25: warning: moving a local object in a return statement prevents copy elision [-Wpessimizing-move]

Source code is

        return std::move(fh);

I think some code rework would make it run faster.

MartyG-RealSense, 2022-08-22:
Hi @dcb314  If yourself or another RealSense user develop an edit of the source code that improves performance then you are very welcome to share it as a **pull request** so that it can be evaluated by the Intel RealSense developer team for inclusion in the RealSense SDK.  Thanks!

https://github.com/IntelRealSense/librealsense/pulls

dcb314, 2022-08-22:
Thanks for your offer. I don't do pull requests.

There were a bunch more of the same warning in the package.

      3 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/common/opengl3.cpp:599:21: warning: moving a local object in a return statement prevents copy elision [-Wpessimizing-move]
     70 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/include/librealsense2/hpp/rs_device.hpp:677:48: warning: moving a temporary object prevents copy elision [-Wpessimizing-move]
     70 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/include/librealsense2/hpp/rs_device.hpp:680:29: warning: moving a temporary object prevents copy elision [-Wpessimizing-move]
      1 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/context.cpp:545:25: warning: moving a local object in a return statement prevents copy elision [-Wpessimizing-move]
      9 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/gl/../../include/librealsense2/hpp/rs_device.hpp:677:48: warning: moving a temporary object prevents copy elision [-Wpessimizing-move]
      9 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/gl/../../include/librealsense2/hpp/rs_device.hpp:680:29: warning: moving a temporary object prevents copy elision [-Wpessimizing-move]
     12 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/../include/librealsense2/hpp/rs_device.hpp:677:48: warning: moving a temporary object prevents copy elision [-Wpessimizing-move]
     12 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/../include/librealsense2/hpp/rs_device.hpp:680:29: warning: moving a temporary object prevents copy elision [-Wpessimizing-move]
      1 fedora/pass297/20220820/librealsense.spec.out:/home/dcb36/rpmbuild/BUILD/librealsense-2.40.0/src/media/ros/ros_reader.cpp:514:25: warning: moving a local object in a return statement prevents copy elision [-Wpessimizing-move]

Column 1 is a count of the number of times the warning occurred in the build.

MartyG-RealSense, 2022-08-22:
This warning has never been previously reported by librealsense users.  Have you experienced it in earlier gcc versions?

dcb314, 2022-08-22:
The warning has recently been enhanced in development versions of gcc.

The most recently released version of gcc (12.2) might produce some of these warnings.

MartyG-RealSense, 2022-08-22:
Let's keep this case open and see whether anybody else reports a similar problem in future with recent development versions of gcc.  Thanks very much for the report!

closed 2023-07-18 by MartyG-RealSense