realsenseai/librealsense #10802: terminate called after throwing an instance of 'rs2::error'   what():  Couldn't resolve requests

zhr1120181282, 2022-08-23:
I am trying to set the parameter of the pipe.
When I start the pipe with default parameter, everything goes well, I can see the deep image and color image.
However, when I start like this, I have this error.
Ubuntu 22.04+D435i.
    rs2::pipeline pipe;
    rs2::config pipe_config;
    pipe_config.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,10);
    pipe_config.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_BGR8,10);

    rs2::pipeline_profile profile = pipe.start(pipe_config);

MartyG-RealSense, 2022-08-23:
Hi @zhr1120181282  The FPS speed of 10 is not supported on the D435i camera model.  Please try 6 or 15.

MartyG-RealSense, 2022-08-30:
Hi @zhr1120181282  Do you require further assistance with this case, please?  Thanks!

zhr1120181282, 2022-08-30:
Hi! It worked, thank you for your assistance!

MartyG-RealSense, 2022-08-30:
You are very welcome.  Thanks very much for the update!  As you achieved a solution, I will close this case.  Thanks again!

closed 2022-08-30 by MartyG-RealSense