realsenseai/librealsense #10803: realsense d435i operation error

zjc25, 2022-08-23:
realsense-viewer
when i execute this instruction    realsense-viewer , it appears
02:10:46 [Info] /home/firefly/librealsense-2.50.0/src/linux/backend-v4l2.cpp:570 - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:8/device/../../../../../../../../../../
23/08 01:59:08,559 WARNING [548215792720] (ds5-factory.cpp:1152) DS5 group_devices is empty.
but i run v4l2-ctl --list-device
rkcif-mipi-lvds2 (platform:rkcif):
	/dev/media0

Intel(R) RealSense(TM) Depth Ca (usb-xhci-hcd.7.auto-1):
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/media1
	/dev/media2

Failed to open /dev/video0: No such device
it have some depth cameras device
so i don`t know what happened

MartyG-RealSense, 2022-08-23:
Hi @zjc25  By default on a D435i camera, /video0 should be the depth stream, whilst /video1 is infrared and /video2 is RGB.

**Failed to read busnum/devnum** is a highly rare error.  In past cases it has usually occurred on very small single-board computers due to insufficient power being provided to the camera if the small computing board has a micro-sized USB port such as **USB 3 OTG (On The Go)** USB port instead of a full-size port.  

Which computer / computing device are you using, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).  Thanks!

zjc25, 2022-08-23:
Hi , @MartyG-RealSense  When i use the cheese in ubuntu , it can work.
So i think it is not about power

zjc25, 2022-08-23:
Hi @MartyG-RealSense The kernel version I use is 5.10.66, and it requires 5.4. Is it related to this?
./script/patch-realsense-ubuntu-lts.sh
When I run this command, it appears,
[image]

MartyG-RealSense, 2022-08-23:
If building the librealsense SDK from packages then the most recent kernel supported is 5.4.

If building the SDK from source code then the most recent kernels currently supported are 5.8 and 5.11.  5.10 is not officially supported, and whilst non-supported kernels can be used with librealsense there may be unpredictable consequences in regards to stability.

An unofficial test patch at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 adds support for kernels 5.13 and 5.15.

Alternatively, if the SDK is built from source code with the **RSUSB backend** installation method that bypasses the kernel then the build is not dependent on Linux versions or kernel versions and does not require patching, so the kernel version being used does not matter.

zjc25, 2022-08-24:
Thank you very much for your answer.The problem that has troubled me for two weeks has finally been solved.
I use 
cmake ../  -DFORCE_RSUSB_BACKEND=ON
Finally,the problem that the camera could not be found was successfully solved
thank you very much

zjc25, 2022-08-24:
Hi @MartyG-RealSense i found a new question
when i run three depth cameras,it has an error
[image]

MartyG-RealSense, 2022-08-24:
**when i run three depth cameras,it has an error**

Does this mean that you are running three cameras at the same time, and it works with two cameras but if you use a third camera you receive this error?

zjc25, 2022-08-25:
yes,but
sometime,it can run three cameras at the same time , but the cpu occupy is very high
Is it caused by kernel mismatch,
[image]

zjc25, 2022-08-25:
i want to know why it appears,and what can i do to reduce its CPU utilization
@MartyG-RealSense

MartyG-RealSense, 2022-08-25:
High CPU utilization from running three cameras simultaneously is not likely related to the kernel version.  The number of cameras that can be simultaneously active on the same computer will depend upon the hardware specification of the computer.  The more cameras that are active, the greater the amount of the computer's resources that will be consumed.  

For 3 cameras I would recommend using a PC with at least an Intel i5 CPU.  If you need to use three cameras with a single-board computer instead of a PC, I would recommend an Nvidia Jetson board.

I note that in https://github.com/IntelRealSense/librealsense/issues/10803#issuecomment-1225086363 you state that you are using **RSUSB backend**.  This backend is best suited to single camera applications rather than multiple-camera setups.  A patch-based build of librealsense (**V4L backend / native backend**) is the ideal for multicam applications.  I acknowledge that you previously had camera detection problems before using RSUSB backend though.

Using a lower color resolution if it is possible for you to do so may help to reduce the CPU usage percentage.

closed 2022-09-06 by MartyG-RealSense