realsenseai/librealsense #10804: What is the official explanation of USB SCP overflow

2513297244, 2022-08-23:


MartyG-RealSense, 2022-08-23:
Hello, in the past a member of the RealSense developer team provided the following explanation of USB SCP overflow.

"The overflows are an indication that the PC is not fetching all frames in time.  Depending on the severity this can cause no streaming (when all frames are lost) or reduction in FPS.  Reducing FPS / resolution should take care of that".

2513297244, 2022-08-23:
> 您好，过去 RealSense 开发团队的成员提供了以下关于 USB SCP 溢出的解释。
> 
> “溢出表明 PC 没有及时获取所有帧。根据严重程度，这可能导致没有流式传输（当所有帧丢失时）或 FPS 降低。降低 FPS / 分辨率应该可以解决这个问题”。

thank you

MartyG-RealSense, 2022-08-30:
Hi, do you require further assistance with this case, please?  Thanks!

2513297244, 2022-08-30:
> 您好，请问您需要对此案的进一步帮助吗？谢谢！

No , thank you!

MartyG-RealSense, 2022-08-30:
Okay, thank you very much!  As you do not require further assistance, I will close this case.  Thanks again!

好的，非常感谢！ 由于您不需要进一步的帮助，我将关闭此案例。 再次感谢！

closed 2022-08-30 by MartyG-RealSense