realsenseai/librealsense #10805: D435 if the frame rate of RGB stream and depth stream is different, will it affect the acquisition of a group of aligned data?

2513297244, 2022-08-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  win10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.50                   | 
| Language                            |  C++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-08-23:
Hello, yes you may see a visible difference in an aligned image that is capturing motion if depth and RGB are at different frames per second (FPS).  For example, if a waving hand is observed by the camera and the FPS speed of the RGB stream is lower than the depth FPS then the RGB data of the aligned image may lag behind the current position of the depth data.

2513297244, 2022-08-23:
> 您好，是的，如果深度和 RGB 处于不同的每秒帧数 (FPS)，您可能会在捕捉运动的对齐图像中看到明显的差异。例如，如果相机观察到挥手，并且 RGB 流的 FPS 速度低于深度 FPS，则对齐图像的 RGB 数据可能落后于深度数据的当前位置。
> 
> 如果相机是静止的并且在观察到的场景中没有发生运动，则差异可能不太明显。

thank you！

MartyG-RealSense, 2022-08-30:
Do you require further assistance with this case, please?  Thanks!

2513297244, 2022-08-30:
> 请问您在这个案子上需要进一步的帮助吗？谢谢！

No,thank you!

MartyG-RealSense, 2022-08-30:
Okay, thank you very much!  As you do not require further assistance, I will close this case.   again!

好的，非常感谢！ 由于您不需要进一步的帮助，我将关闭此案例。 再次感谢！

closed 2022-08-30 by MartyG-RealSense