realsenseai/librealsense #10808: Announcement: RealSense SDK 2.0 (librealsense2) version 2.51.0 now available

MartyG-RealSense, 2022-08-23:
Hi everyone,

Version **2.51.0** of the RealSense SDK 2.0 (librealsense2) is now available at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

**Release Notes**
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2511

kliver, 2022-08-23:
Why was Android official support removed and what does it means for people developing intel realsense applications for Android devices? we will not longer get updates and support?

ThatGeoGuy, 2022-08-23:
Question: Why is this pushed from the `development` branch instead of `master`? If you look at the repository from the default branch it makes it seem as if no development has happened since November 2021.

MartyG-RealSense, 2022-08-24:
Hi @ThatGeoGuy  The **master** GitHub branch's history reflected the November 2021 history yesterday but has now updated to the current 2.51.1 history.

[image]

MartyG-RealSense, 2022-08-24:
Hi @kliver  There is the option of continuing to use SDK version **2.50.0** if you are concerned about the removal of official support for Android and MacOS from 2.51.1 onwards.

The Android wrapper currently supports up to **Android 11**.  RealSense users may be able to create a custom GitHub 'fork' version of the open source librealsense SDK and add their own support for newer Android versions.

jb455, 2022-08-25:
Hi, can you explain more about dropping Android support please? Is there any official announcement to be made?
The wrapper code is still in the master branch after it's updated to 2.51.1 - is the code going to remain but just not get updated any more to support new products or fix bugs etc?
Is there any particular reason for dropping it? Are there issues with latest Android that you don't want to or can't fix?

Thanks.

closed 2022-08-29 by MartyG-RealSense