realsenseai/librealsense #10810: camera realsense sr305

khangTi, 2022-08-24:
i am using realsense realsense SR305 camera running on android device, i am running the resolution is HD, i set the color image to 1280x720 and the depth image is 640x480 but many times when i turn on the camera, the 2 frames are not equal in size to each other. , i think if it will align to the same size of 2 frames, how to solve this problem admin
thanks admin

MartyG-RealSense, 2022-08-24:
Hi @khangTi  There is an example of Android depth-color alignment code at https://github.com/IntelRealSense/librealsense/issues/4716 that you can check your own program against.

khangTi, 2022-08-25:
Thank you very much admin, I have solved this error

MartyG-RealSense, 2022-08-25:
The SDK's **Camera** Android tool has an example of enableRecordToFile code in its **RecordingActivity.java** script

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/tools/camera/src/main/java/com/intel/realsense/camera/RecordingActivity.java#L100

The 'Capture' tool is the open-source code of the pre-built **RS Camera** Android app on the Google Play store.

https://play.google.com/store/apps/details?id=com.intel.realsense.camera&hl=en_IE&gl=US

closed 2022-09-06 by MartyG-RealSense