realsenseai/librealsense #10815: Cant install realsense-viewer?

MyOtherNamesWereTaken, 2022-08-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | ( | 
| Operating System & Version |   Ubuntu 22.04.1 LTS 
| Kernel Version (Linux Only)    |  5.15.0-46-generic                                    | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
After upgrading to ubuntu 22.04.1 LTS i dont seem to have realsense-viewer installed anymore and cant do so either? Also when trying to run any realsense related code im getting the following error 

RuntimeError: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.5/0003:8086:0B3A.0005/HID-SENSOR-200076.4.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied

What do?

MartyG-RealSense, 2022-08-24:
Hi @MyOtherNamesWereTaken  The librealsense SDK does not currently have official support for Ubuntu 22.04 (Jammy) or kernel 5.15.  An unofficial test patch for kernel 5.13 and 5.15 support and a test package is available at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

https://github.com/IntelRealSense/librealsense/issues/10792#issuecomment-1220320263 provides advice for installing the test package.

MyOtherNamesWereTaken, 2022-08-24:
When will the official support be released?

MartyG-RealSense, 2022-08-24:
Ubuntu 22.04 Jammy support is being worked on by Intel, though I do not have a time schedule to provide for formal implementation at the time of writing this.  I do apologize.

closed 2022-08-31 by MyOtherNamesWereTaken