realsenseai/librealsense #10819: Patching Kernel Modules with new jetpack release (5.0.2)

augustelalande, 2022-08-24:
With the production release of jetpack 5.0.2 I upgraded our jetson machine and am now trying to install librealsense. Following the [instructions](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_jetson.md), I get stuck at the patching stage. Running the command
`./scripts/patch-realsense-ubuntu-L4T.sh`

I get the printout

```
The script patches and applies in-tree kernel modules required for Librealsense SDK

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.5).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Jetson Board (proc/device-tree/model): Jetson-AGX
Jetson L4T version: 35.1
Unsupported JetPack revision 35.1 aborting script
```

It's understandable that the script hasn't been upgraded to support the latest release, but is this something we can expect soon? How difficult will it be to make this run on the new release?

MartyG-RealSense, 2022-08-24:
Hi @augustelalande  The comment at https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1225443624 and my response beneath it should be a helpful reference as it covers the same problem.

augustelalande, 2022-08-24:
Ok thanks. Any progress on development since that comment?

MartyG-RealSense, 2022-08-25:
There is not further progress to report yet at the time of writing this.

MartyG-RealSense, 2022-09-06:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-09-06 by MartyG-RealSense

augustelalande, 2023-06-07:
Awesome @MartyG-RealSense. Appreciate the update.