realsenseai/librealsense #10820: Python bindings don't respect CMAKE_INSTALL_PREFIX

m-elwin, 2022-08-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Any | 
| Firmware Version                   | N/A} | 
| Operating System & Version |   {Linux}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.50.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When configuring a build with -DBUILD_PYTHON_BINDINGS=ON, the generated
python libraries are always installed relative to / (e.g., /usr/lib/python3.10/dist-packages), regardless of the value of CMAKE_INSTALL_PREFIX. 

Specifying -DPYTHON_INSTALL_DIR manually works around this issue.

I believe the issue is related to how the Python install directory is determined.
In wrappers/python/CMakeLists.txt:64, PYTHON_INSTALL_DIR is set to ${Python_SITEARCH}, which I believe is always an absolute path.

I think the code used for CMake Version <3.12 may not have this issue.

MartyG-RealSense, 2022-08-25:
Thanks very much @m-elwin for highlighting this CMake issue to the RealSense community and advising about a manual workaround!

MartyG-RealSense, 2022-09-01:
Case closed due to no further comments received.  Thanks again @m-elwin for sharing your workaround!

closed 2022-09-01 by MartyG-RealSense

maloel, 2022-09-21:
@m-elwin 
First, thanks for your contribution.
But I have to ask, why do you want to override the install directory?
Installation is supposed to put the compiled code in a place where a python `import` command would see it. Python itself sets this folder and, according to the [docs](https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/#_mandatory_macros), that's the folder that should be used.

Do you have two different Python installations? Is CMake finding the wrong Python executable?

m-elwin, 2022-09-27:
Wanted to install under my home directory instead of system python directory (which requires root).