realsenseai/librealsense #10821: opencv rotate-pointcloud example crash

Sylvexia, 2022-08-25:
| Required Info                         | opencv rotate-pointcloud example crash |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   L515  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Ubuntu 20  | 
| Kernel Version (Linux Only)    | jetpack 5.0.2 | 
| Platform                                 |NVIDIA Jetson NX module 16GB|
| SDK Version                            | 2 | 
| Language                            | c++ | 
| Segment			|Robot| 

### Issue Description
In wrappers/opencv/rotate-pointcloud example
The pixel showed on window seems doesn't register correctly.
And crash with no sign.
The error does not show the same every time it cashed.
But they are all about the memory issue
For instance:

```
free(): invalid next size (normal)
Aborted (core dumped)
```
```
malloc(): corrupted top size
Aborted (core dumped)
```

MartyG-RealSense, 2022-08-25:
Hi @Sylvexia  I would recommend commenting out the threshold filter post-processing filter in the script, as it is not supported on the L515 camera model.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/rotate-pointcloud/rs-rotate-pc.cpp#L61-L68

Does the program become more stable if these lines are commented out, please?

MartyG-RealSense, 2022-08-25:
It's great to hear that you were successful!

The majority of C++ code designed for 400 Series cameras will work with L515 too.  The main exceptions will be in the following functions:

*  Post-processing filters: only **Temporal** for depth and **Decimation** for RGB are supported on L515.

*  Visual Presets: the L515 has a different set of built-in camera configuration presets to the L515

*  Resolution and FPS: only 30 FPS is supported for depth, and only the resolutions 320x480, 640x480 and 1024x768 are supported for depth.  In scripts that set 848x480 or 1280x720 as the depth resolution or use an FPS other than 30, you would need to edit those script values to ones supported on L515.

Sylvexia, 2022-08-25:
Thanks for the help!

closed 2022-08-25 by Sylvexia

MartyG-RealSense, 2022-08-25:
You are very welcome.  I'm pleased that I could help!