realsenseai/librealsense #10823: unity and D435i IMU ?

aelectr, 2022-08-25:
hi

I am trying to save the point cloud as mesh in unity. The generated point cloud disappears as the camera moves to the other angle. Also what settings work to stream imu position and rotation to unity?

MartyG-RealSense, 2022-08-25:
Hi @aelectr  Does the point cloud disappear only when the camera is being moved and then reappear when the camera stops moving, please?

The image below provides an example of configuring RsDevice profiles in Unity for the IMU streams (Accel and Gyro).

[image]

aelectr, 2022-08-25:
Yes, point clouds only appear at the angle where the camera faces, so for constructing the environment using the point cloud, it is necessary that the point clouds don't disappear once we turn the camera around.
I also implement the settings, and this error comes up:

rs2_pipeline_start_with_config(pipe:00000238B6AABD40, config:00000238B6AAC200)

aelectr, 2022-08-26:
And could you help me with how I can show the stream data on canvas? How can I read the stream data?

MartyG-RealSense, 2022-08-26:
The supported pairings for the Accel and Gyro FPS are Accel 63 / Gyro 200, and Accel 250 / Gyro 400.

The RealSense Unity wrapper converts camera data into a Unity material in order to render it, and so textual / numeric values cannot be displayed onscreen as a part of that rendered material.

aelectr, 2022-08-26:
I want to show the trajectory of the camera inside unity when it moves, so based on what you said isn't possible?

aelectr, 2022-08-26:
@MartyG-RealSense , Is there any RGBD SLAM work that has been done in unity using the theRealSense Unity SDK and D435i?

aelectr, 2022-08-26:
@MartyG-RealSense 
[image]
This is what I want to achieve within unity, the example works fine in ROS but I am looking to save the cloud points while scanning the entire area but unfortunately, the texture disappears and no longer save in-game or editor view.

MartyG-RealSense, 2022-08-26:
I do not know of a Unity SLAM solution specifically for D435i.  

The RealSense SDK does have a Unity example SLAM program for the **T265 Tracking Camera** model only.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scenes/Samples/SLAM.unity

https://www.youtube.com/watch?v=Ea_e1iVX7LU

***************

While not specific to RealSense, Unity have an official video tutorial about performing SLAM using ROS2 at the link below.

https://resources.unity.com/automotive-transportation-manufacturing-content/unity-robotics-slam

Also by Unity:

https://blog.unity.com/manufacturing/introducing-unity-robotics-visualizations-package

https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example

closed 2022-09-07 by MartyG-RealSense