realsenseai/librealsense #10825: Play Pause Stop Resume

Zumbalamambo, 2022-08-25:
How do I play pause stop and resume in Unity SDK?

MartyG-RealSense, 2022-08-26:
Hi @Zumbalamambo  At https://github.com/IntelRealSense/librealsense/issues/2635#issuecomment-435588827 there is a RealSense Unity script shared by a RealSense user that demonstrates use of pause() and resume()

Zumbalamambo, 2022-08-26:
@MartyG-RealSense thank you but I want to play/pause the live stream and not the playback stream?

MartyG-RealSense, 2022-08-27:
My understanding is that live streams cannot be paused, as a RealSense team member advises at https://github.com/IntelRealSense/librealsense/issues/1527#issuecomment-381413924

Though a live stream can appear to be paused in the RealSense Viewer by pressing the space bar key or the pause icon, it is not a true pause, as if a bag file recording is being made at the time then the stream continues recording during the pause and has no pauses on the playback.

MartyG-RealSense, 2022-09-02:
Hi @Zumbalamambo  Do you require further assistance with this case, please?  Thanks!

Zumbalamambo, 2022-09-02:
@MartyG-RealSense It will be useful to have this feature. Until then, i'll enable/disable the entire game object to play/stop.

closed 2022-09-07 by Zumbalamambo