realsenseai/librealsense #10828: Build Intel RealSense SDK by cmake.

qscesz19970810, 2022-08-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | 0.5.15.00.50 | 
| Operating System & Version |   {Linux-Android}  | 
| Kernel Version (Linux Only)    | 4.19.133 | 
| Platform                                 | automotive Android 10  |
| SDK Version                            |  2.5.0 | 
| Language                            |  C/C++ }                          | 
| Segment			|  {others } | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello,i want to run realsense on Android,but I just compiled the code into an executable and pushed into the device /system/bin.

My device gained root accesss,so i follow this doc(https://dev.intelrealsense.com/docs/build-intel-realsense-sdk-headless-tools-and-examples) to do.

Beacause i don't find example.h and rs-depth in new version librealsense,so i download this document(https://github.com/IntelRealSense/librealsense/blob/master/examples/C/depth/rs-depth.c) to compiler the executable.

This is my cmakelists:
-------
cmake_minimum_required(VERSION 3.4.1)
project(main)

add_library(realsense2 STATIC IMPORTED)
set_target_properties(
    realsense2
    PROPERTIES
    IMPORTED_LOCATION
    ~/librealsense/build/librealsense2.so
)

add_library(log-lib STATIC IMPORTED)
set_target_properties(
    log-lib
    PROPERTIES
    IMPORTED_LOCATION
    ~/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so
)

add_library(m-lib STATIC IMPORTED)
set_target_properties(
    m-lib
    PROPERTIES
    IMPORTED_LOCATION
    ~/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/21/libm.so
)

add_executable(main main.c)

target_link_libraries(main
    log-lib
    realsense2
    m-lib
    )
-----

When i run the executable on my root device,just get "There are 0 connected Realsense devices".

MartyG-RealSense, 2022-08-26:
Hi @qscesz19970810  Did you build the RealSense Android wrapper using the instructions at the link below, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android

In the original Android wrapper several years ago it was necessary to root the Android device to gain access to it.  A later wrapper (the one at the link above) enabled **unrooted** access.

Android 10 also has some compatibility issues with the Android wrapper where you may need to set **targetSdkVersion** to a value of '27' in order for the wrapper to work with an Android 10 device.  Android 11 does not have this issue.

qscesz19970810, 2022-08-29:
hello,i don't use Android Studio IDE. i wish use cmake to build librealsense2.so file.

my device have gain root access,so i follow this https://dev.intelrealsense.com/docs/build-intel-realsense-sdk-headless-tools-and-examples to build file.

MartyG-RealSense, 2022-08-29:
If you are cloning the master branch of the RealSense SDK (step 5 of the headless tool instructions) then the **example.h** file and the project folder for rs-depth should be at the below location in your cloned SDK:

https://github.com/IntelRealSense/librealsense/tree/master/examples/C

[image]

I believe that step 6 is suggesting that you then edit the rs-depth and rs-distance files within your cloned SDK instead of downloading them separately from the original librealsense GitHub.

*******

It has been several years since the rooted Android wrapper has been in use due to the unrooted wrapper superseding it, so available information about using the rooted wrapper is very limited.

qscesz19970810, 2022-08-30:
Ok,Thanks

When i use command "cmake .. -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=<Ndk_path>/build/cmake/android.toolchain.cmake -DFORCE_LIBUVC=TRUE" have a question.

The compiler setting is clang/clang++,not is cross compiler aarch64-linux-android29-clang/aarch64-linux-android29-clang++.

Is the new version not supported, or do I need to set something else?

Thank you for your help @MartyG-RealSense

MartyG-RealSense, 2022-08-30:
Are you asking whether the rooted wrapper supports Android SDK 29, please?  If so then there is no available information about that due to the significant age of that wrapper.

The non-detection of the RealSense camera indicated by the **"There are 0 connected Realsense devices"** message makes me think though that it is either a permissions-related problem or an issue with the OTG cable that you are using.  It is my understanding that not all OTG cables / adaptors are equal in regard to how well they work with RealSense cameras.

qscesz19970810, 2022-08-31:
I confirm the OTG cable is ok.
When i use cmake to build the librealsense2.so,generates many warningshe.
I'm wondering the librealsense2.so is incomplete,but I don't know how to solve the problem
So I want check cmake method is still used realsense sdk 2.5.0>

mds1@mds1-ROG-Strix-G533ZM-G533ZM:~/Code/librealsense/build$ cmake .. -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=/home/mds1/CrossCompiler/android-ndk-r21-linux-x86_64/android-ndk-r21/build/cmake/android.toolchain.cmake -DFORCE_RSUSB_BACKEND=ON -DANDROID_PLATFORM=android-29 -DCHECK_FOR_UPDATES=OFF
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mds1/CrossCompiler/android-ndk-r21-linux-x86_64/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mds1/CrossCompiler/android-ndk-r21-linux-x86_64/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.51.1
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Android configurations.The Wainning gain by cmake/android_config.cmake.
-- Prepare RealSense SDK for Android OS (arm64-v8a)
-- using RS2_USE_ANDROID_BACKEND
-- Prepare RealSense SDK for Android OS (arm64-v8a)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mds1/Code/librealsense/build
mds1@mds1-ROG-Strix-G533ZM-G533ZM:~/Code/librealsense/build$ make
[  0%] Building C object third-party/realsense-file/CMakeFiles/realsense-file.dir/lz4/lz4.c.o
[  1%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/console_bridge/src/console.cpp.o
[  1%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/debug.cpp.o
[  2%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/cpp_common/src/header.cpp.o
In file included from /home/mds1/Code/librealsense/third-party/realsense-file/rosbag/cpp_common/src/header.cpp:37:
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:79:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_ERROR, fmt, ##__VA_ARGS__)
                                                                                         ^
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:79:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
2 warnings generated.
[  2%] Building CXX object third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag.cpp.o
In file included from /home/mds1/Code/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/bag.cpp:28:
In file included from /home/mds1/Code/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/bag.h:64:
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/../../../console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_DEBUG, fmt, ##__VA_ARGS__)
                                                                                         ^
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/../../../console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/home/mds1/Code/librealsense/third-party/realsense-file/rosbag/console_bridge/include/console_bridge/console.h:88:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-ma [... cut ...]
[ 97%] Building CXX object CMakeFiles/realsense2.dir/src/points.cpp.o
[ 97%] Building CXX object CMakeFiles/realsense2.dir/src/to-string.cpp.o
[ 98%] Building C object CMakeFiles/realsense2.dir/third-party/sqlite/sqlite3.c.o
[ 98%] Building CXX object CMakeFiles/realsense2.dir/third-party/easyloggingpp/src/easylogging++.cc.o
[ 99%] Building CXX object CMakeFiles/realsense2.dir/common/utilities/time/l500/get-mfr-ww.cpp.o
[100%] Linking CXX shared library librealsense2.so
[100%] Built target realsense2

MartyG-RealSense, 2022-08-31:
I note that the CMake command for building the SDK is:

`cmake .. -DANDROID_ABI=<Application Binary Interface> -DCMAKE_TOOLCHAIN_FILE=<Path to NDK folder>/build/cmake/android.toolchain.cmake -DFORCE_LIBUVC=TRUE`

It looks as though it is expected that the parts of the instructions in brackets like <Application Binary Interface> and <Path to NDK folder> are supposed to be replaced with the configuration details of your particular Android device and filepath.  Examples of configurations are provided at https://github.com/IntelRealSense/librealsense/issues/5831#issuecomment-585761960 and https://github.com/IntelRealSense/librealsense/issues/2061

MartyG-RealSense, 2022-09-06:
Hi @qscesz19970810  Do you require further assistance with this case, please?  Thanks!

qscesz19970810, 2022-09-12:
I tried it and it still doesn't work.

At android native_example,I see line 17 "private RsContext mRsContext;" and line 38 "RsContext.init(getApplicationContext());"

Can I be sure about the content of this step,It doesn't work when I'm missing this command

I'm wondering if I'm missing this step in my C++ program so it doesn't work.

"RsContext.init" <- Is there any replace command in c++? i saw context not have init() attributes

qscesz19970810, 2022-09-13:
Yes, what I'm asking is if I run with an executable file(Just have C/C++ code)
Does not do this initialization have any effect?
If yes, do I have any replace command to use? Because C++ "rs::context: does not have the attribute of init().

MartyG-RealSense, 2022-09-13:
The only kind of replace command that I know of in C++ is string replace.

https://www.javatpoint.com/cpp-string-replace-function

A RealSense team member advises at https://github.com/IntelRealSense/librealsense/issues/3385#issuecomment-469712895 that it is possible to build librealsense2.so for the current unrooted Android wrapper with CMake if a dedicated CMake flag is set   One of the developers of the Android wrapper provides further advice about permissions in that discussion at https://github.com/IntelRealSense/librealsense/issues/3385#issuecomment-481723287

The article at the link below may also be worth reading.

https://www.anycodings.com/1questions/1610107/building-librealsense-android-app-with-ndk-native-c-code-part

I do not know the answer to your questions about running with an executable file though, unfortunately.  I do apologize.

qscesz19970810, 2022-09-16:
Thanks,my question is the UVC settings are not written.....i got that.
I'm not asking about string repalace XD

closed 2022-09-16 by qscesz19970810