realsenseai/librealsense #10829: Realsense d435i is not detected on docker

goyalvarun02, 2022-08-26:
I am using Ubuntu 22.04 host machine and created a VM of ubuntu 18.04 to use Melodic version of ROS with Realsense camera. 

But I am using docker on VM (ubuntu 18.04) , and when I tried to run 
roslaunch realsense2_camera rs_camera.launch

I got an error :
"Failed to set power state" as shown in pic
 
I am new to librealsense please advise
[image]

goyalvarun02, 2022-08-26:
[image]
Copied the rules as well.

MartyG-RealSense, 2022-08-26:
Hi @goyalvarun02  Is the camera detected if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

in regard to the rs-enumate-devices command that you attempted, the command is:

**rs-enumerate-devices**

goyalvarun02, 2022-08-26:
Hi @MartyG-RealSense ,

PFA image of the output after running 
roslaunch realsense2_camera rs_camera.launch initial_reset:=true
[image]
[image]

Pls advice

MartyG-RealSense, 2022-08-26:
Can you confirm please that you used initial_reset, as your log image shows initial-reest

goyalvarun02, 2022-08-26:
Hi @MartyG-RealSense  I

I am really sorry for typo error .
PFA logs

[image]

[image]

MartyG-RealSense, 2022-08-26:
No problem at all  :)  Which method did you use to install the librealsense SDK and the RealSense ROS wrapper?  Did you install them separately (the SDK first and the wrapper second) or install them both at the same time from packages with the **Method 1** instructions in the ROS wrapper?

goyalvarun02, 2022-08-26:
I first installed ROS-Melodic and then use command 
sudo apt-get install ros-$ROS_DISTRO-realsense2-camera
sudo apt-get install ros-$ROS_DISTRO-realsense2-description

thereafter created a catkin_ws with clone of repository of xarm_ros and realsense . Compiled the realsense in Catkin_src and install.

MartyG-RealSense, 2022-08-26:
Using **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** is fine.  This will install both the librealsense SDK and the ROS wrapper from packages.

You will not likely need to install **realsense2-description** unless you are performing ROS robot simulations such as Gazebo.

Creating a catkin_ws with a clone of RealSense sounds as though it has the potential to cause a conflict with the package installation that you made before taking that step.  If you have installed from packages then you should not then need to install from source with cloning.

goyalvarun02, 2022-08-26:
Okay So I can remove the realsense-ros repo , recompiled and see if there is no conflict? 
Is that ok?

MartyG-RealSense, 2022-08-26:
Does the error still occur if you delete the entire **catkin_ws/src** folder?

goyalvarun02, 2022-08-26:
I can not delete the entire SRC folder but  have deleted the build and devel folder to clean recompiled. But still error occurred.

MartyG-RealSense, 2022-08-26:
My recollection is that it is advised that only RealSense files should be installed in the catkin_ws workspace and not non-RealSense files.

Is the camera able to be detected if you run the **realsense-viewer** tool in the Ubuntu terminal?

goyalvarun02, 2022-08-26:
For other setup I have kept the non-Realsense files as well in catkin_ws it was working fine . 

I did not find the command realsense-viewer in ubuntu ? Do I need to install anything for it.

MartyG-RealSense, 2022-08-26:
I'm not certain whether the realsense-viewer tool is included in a librealsense build created with the ROS wrapper's apt-get installation instructions.

What details are listed if you do **rs-enumerate-devices**

goyalvarun02, 2022-08-26:
rs-enumerate-devices command not found .

I am trying to install in docker but still looking for it to install.

MartyG-RealSense, 2022-08-26:
Which VM tool are you using, please?  Intel recommend VMWare Workstation Player (and not Oracle VrtualBox) because it provides proper emulation of the USB 3 controller.

goyalvarun02, 2022-08-26:
I am using oracle VM and also done the enabling of Virtualization and USB 3 in Bios setting .

MartyG-RealSense, 2022-08-26:
My knowledge of Docker setups is admittedly limited.  Also, Ubuntu 22.04 is not yet officially supported by librealsense - there is an unofficial test installation at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 - though my understanding is that the host machine's Ubuntu version should not affect the librealsense build on the VM's Ubuntu that us using 18.04.   I am also not familiar with xarm_ros.

What details are shown if you input **lsusb** in the Ubuntu terminal?  lsusb is an Ubuntu tool rather than a RealSense one and so should work no matter what problems a RealSene installation has.

goyalvarun02, 2022-08-29:
Hi @MartyG-RealSense  , 

When I do lsusb in host machine/ VM /Docker it shows intel real-sense is detected and USB 007 is used for it. 

I am struggling and could not find the root cause of this issue .
Please advice.

goyalvarun02, 2022-08-30:
Hi @MartyG-RealSense 

I observed following things:
1. When I run the cheese app in ubuntu , camera is able to show Videos (and does not on some machine I observed random behavior) .
2. However, when I run the command in docker 
roslaunch realsense2_camera rs_camera.launch initial_reset:=true
I observed the power error (as shown in pic) . But I can still see the output of "lsusb" in docker and can see device is identified .

PFA image
[image]

Seeking your help to fix this issue

goyalvarun02, 2022-08-30:
Hi @MartyG-RealSense 
I observerd wiered behaviour , first it detect the camera later on it says disconnected (without doing anything)

PFA full logs as well . 

root@ubuntu-22:/# roslaunch realsense2_camera rs_camera.launch 
bash: roslaunch: command not found
root@ubuntu-22:/# source catkin_ws/devel/setup.bash 
root@ubuntu-22:/# rs-enumerate-devices
bash: rs-enumerate-devices: command not found
root@ubuntu-22:/# roslaunch realsense2_camera rs_camera.launch 
... logging to /root/.ros/log/8d092cf0-283a-11ed-a696-6cb3115d58d8/roslaunch-ubuntu-22-75.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-22:44361/

SUMMARY
========
PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 137 lines cut ...]
 30/08 08:06:15,576 WARNING [140408696850176] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1661846781.604508143]:  
[ WARN] [1661846781.633985287]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1661846781.634049163]: The requested device with  is NOT found. Will Try again.
 30/08 08:06:21,633 ERROR [140409103279872] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 30/08 08:06:21,633 ERROR [140408696850176] (sensor.cpp:572) acquire_power failed: failed to set power state
 30/08 08:06:21,633 WARNING [140408696850176] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1661846787.661767565]:  
[ WARN] [1661846787.688535543]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1661846787.688601885]: The requested device with  is NOT found. Will Try again.
 30/08 08:06:27,687 ERROR [140409120065280] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 30/08 08:06:27,687 ERROR [140408696850176] (sensor.cpp:572) acquire_power failed: failed to set power state
 30/08 08:06:27,688 WARNING [140408696850176] (rs.cpp:310) null pointer passed for argument "device"
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@ubuntu-22:/# 

 .
 .
 .

MartyG-RealSense, 2022-08-30:
Are pre-built executable versions of RealSense examples and tools such as rs-enumerate-devices and realsense-viewer present in the **usr/local/bin** folder of Ubuntu 18.04 after you installed the SDK and wrapper from packages with **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** please?

Launch commands for these examples and tools should be input into the Ubuntu terminal and not the ROS terminal.

As the roslaunch proceeds normally with the camera detected up until the **RealSense Node is Up!** message and then is no longer detected, it does again suggest a power related issue (the camera disconnecting as the power demands of the camera increase as the streams start publishing).  Because you can output RGB from cheese, this suggests that the depth stream might be the cause of the **Failed to set power state** warnings.

Does it still disconnect if you add **enable_depth:=false** to your launch instruction so that it only publishes color?

goyalvarun02, 2022-08-31:
Hi @MartyG-RealSense 

1. I have checked /usr/local/bin folder for both working realsense2-camera setup (assume setup 1, where command works fine) and non working realsense-camera(assume setup 2, where command is not found). I observed in both rs-enumerate -device and realsense-viewer is not present. 

Setup1 :
[image]

Setup 2:
[image]

2. I run the roslaunch with enable_Depth:=false and output is same as earlier:
[image]
[image]

MartyG-RealSense, 2022-08-31:
Let's try installing from packages only the examples and tools, not the librealsrnse SDK (which is already installed)

1.  Register the server's public key

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`

2.  Add the server to the list of repositories:

`sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`

3.  Install only the examples and tools

`sudo apt-get install librealsense2-utils`

goyalvarun02, 2022-08-31:
Hi @MartyG-RealSense .

Here is the output of commands:

1. Output of rs-enumerate-devices
31/08 05:47:13,551 ERROR [139729328211648] (librealsense-exception.h:52) Cannot identify '/dev/video14 Last Error: No such file or directory
 31/08 05:47:13,568 ERROR [139729328211648] (librealsense-exception.h:52) Cannot identify '/dev/video14 Last Error: No such file or directory
 31/08 05:47:13,574 ERROR [139729328211648] (librealsense-exception.h:52) Cannot identify '/dev/video14 Last Error: No such file or directory
 31/08 05:47:13,579 ERROR [139729328211648] (librealsense-exception.h:52) Cannot identify '/dev/video14 Last Error: No such file or directory
 31/08 05:47:13,591 ERROR [139729328211648] (librealsense-exception.h:52) Cannot identify '/dev/video14 Last Error: No such file or directory
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	817512070833
    Firmware Version              : 	05.11.06.250
    Recommended Firmware Version  : 	05.13.00.50
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	D400
    Asic Serial Number            : 	814113021595
    Firmware Update Id            : 	814113021595

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1280x720	@ 6Hz	   RGB8
    Color	  1280x720	@ 6Hz	   Y16
    Color	  1280x720	@ 6Hz	   BGRA8
    Color	  1280x720	@ 6Hz	   RGBA8
    Color	  1280x720	@ 6Hz	   BGR8
    Color	  1280x720	@ 6Hz	   YUYV
    Color	  640x480	@ 30Hz	   RGB8
    Color	  640x480	@ 30Hz	   Y16
    Color	  640x480	@ 30Hz	   BGRA8
    Color	  640x480	@ 30Hz	   RGBA8
    Color	  640x480	@ 30Hz	   BGR8
    Color	  640x480	@ 30Hz	   YUYV
    Color	  640x480	@ 15Hz	   RGB8
    Color	  640x480	@ 15Hz	   Y16
    Color	  640x480	@ 15Hz	   BGRA8
    Color	  640x480	@ 15Hz	   RGBA8
    Color	  640x480	@ 15Hz	   BGR8
    Color	  640x480	@ 15Hz	   YUYV
    Color	  640x480	@ 6Hz	   RGB8
    Color	  640x480	@ 6Hz	   Y16
    Color	  640x480	@ 6Hz	   BGRA8
    Color	  640x480	@ 6Hz	   RGBA8
    Color	  640x480	@ 6Hz	   BGR8
    Color	  640x480	@ 6Hz	   YUYV
    Color	  424x240	@ 60Hz	   RGB8
    Color	  424x240	@ 60Hz	   Y16
    Color	  424x240	@ 60Hz	   BGRA8
    Color	  424x240	@ 60Hz	   RGBA8
    Color	  424x240	@ 60Hz	   BGR8
    Color	  424x240	@ 60Hz	   YUYV
    Color	  424x240	@ 30Hz	   RGB8
    Color	  424x240	@ 30Hz	   Y16
Color	  424x240	@ 30Hz	   BGRA8
    Color	  424x240	@ 30Hz	   RGBA8
    Color	  424x240	@ 30Hz	   BGR8
    Color	  424x240	@ 30Hz	   YUYV
    Color	  424x240	@ 15Hz	   RGB8
    Color	  424x240	@ 15Hz	   Y16
    Color	  424x240	@ 15Hz	   BGRA8
    Color	  424x240	@ 15Hz	   RGBA8
    Color	  424x240	@ 15Hz	   BGR8
    Color	  424x240	@ 15Hz	   YUYV
    Color	  424x240	@ 6Hz	   RGB8
    Color	  424x240	@ 6Hz	   Y16
    Color	  424x240	@ 6Hz	   BGRA8
    Color	  424x240	@ 6Hz	   RGBA8
    Color	  424x240	@ 6Hz	   BGR8
    Color	  424x240	@ 6Hz	   YUYV

2. realsense-viewer
[image]

[image]

Output of roslaunch command is still same :-(

MartyG-RealSense, 2022-08-31:
It is good that realsense-viewer is working for you now.  Your image of it highlights a couple of issues though.

1.  Your currently installed camera firmware driver version is shown as **5.11.6.250**.  This is severely out of date as it was first released in June 2019 and there have been 21 newer firmware versions released since then.

The recommended firmware version for use with SDK 2.51.0 is **5.13.0.50**.

2.  The red warning **Multiple realsense udev-rules were found!** indicates that you have more than one version of the **udev** Ubuntu device handling rules installed on your computer, which can cause conflicts.

You can correct this easily by going to the location below on your computer and deleting the **/etc** udev rule there, and then the error should disappear.

**/etc/udev/rules.d/99-realsense-libusb.rules**

goyalvarun02, 2022-09-01:
Hi @MartyG-RealSense 

I have deleted the rules from /etc/udev/rules.d/99* and updated the firmware.
But after firmware update no device found :-( 

After firmware update when I observed error , I reconnect the camera again it says need to update firmware , Why so ? May I know the reason If I have update firmware once it should be persistent.

PFA images for your reference (output of lsusb, rs-enumerate device after firmware updates)
[image]
(
[image]
)
[image]

[image]
[image]
[image]

MartyG-RealSense, 2022-09-01:
When using the **Update Firmware** option in the Viewer, a firmware bin file should first be downloaded to your computer from the list at the link below.  The most recent firmware at the time of writing this, 5.13.0.50, is at the top of the list.

Once the bin file is downloaded then when the Update Firmware option is selected, navigate the file selector pop-up window to the file at the computer location that it was downloaded to and select the file.  The firmware update should then begin automatically.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

[image]

It is recommendable that the firmware update is performed using the official 1 meter long USB cable supplied with the camera if possible.

Your kindly provided images indicate to me that a firmware update commenced and reached the 30% point.  At this point, the camera resets itself, disconnects for a moment and then is re-detected and continues the firmware installation process.  In a small number of cases though, the camera may not be detected again after the reset at the 30% point and the firmware update fails.

Are you performing the firmware update outside of Docker, please?

Because the firmware is stored inside the camera hardware instead of on the computer, it is possible to perform the update on another computer if you have one available and then take the camera back to the original computer with the updated firmware inside it.

MartyG-RealSense, 2022-09-07:
Hi @goyalvarun02  Do you require further assistance with this case, please?  Thanks!

goyalvarun02, 2022-09-07:
Hi @MartyG-RealSense  ,

After upgrading to latest FW , it works and I did not observer " Power state failure error" .

Thanks for help.

MartyG-RealSense, 2022-09-07:
That's great to hear, @goyalvarun02 - thanks very much for the update!

closed 2022-09-13 by MartyG-RealSense