realsenseai/librealsense #10830: Low Frame Rate  of C#

jinneigeyu, 2022-08-26:
i'm using .net sdk (C# wrappers) , but i found  sensor frame rate was too low ，how to fix this problem

MartyG-RealSense, 2022-08-26:
Hi @jinneigeyu  Could you provide more information about what your C# project does, please?

Also, which computer or computing device are you using (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)

The easiest first step to take when seeking to improve frame rate is to reduce resolution or frames per second (FPS) speed if it possible for your project to do so.

jinneigeyu, 2022-08-26:
here is the code form C# examples and cpp example of rs-capture
`      public FrameSet WaitForFrames(uint timeout_ms = 5000)
        {
            object error;
            var start = DateTime.Now;
            var ptr = NativeMethods.rs2_pipeline_wait_for_frames(Handle, timeout_ms, out error); 
            Console.WriteLine($"{(DateTime.Now - start).TotalMilliseconds} ms a cycle");
            return FrameSet.Create(ptr);
        }
`
**console  output:**
Using device 0, an Intel RealSense D455
    Serial number: 146222253912
    Firmware version: 05.13.00.50
1577.4387 ms a cycle
162.7369 ms a cycle
206.6041 ms a cycle
207.007 ms a cycle
207.577 ms a cycle
208.1215 ms a cycle
208.9984 ms a cycle
206.02 ms a cycle
207.0063 ms a cycle
208.4333 ms a cycle
207.202 ms a cycle

-----------------------------------------
but in cpp project: 
` while (app) // Application still alive?
    {
        auto start = std::chrono::high_resolution_clock::now();
        
        rs2::frameset data = pipe.wait_for_frames().    // Wait for next set of frames from the camera
                             apply_filter(printer).     // Print each enabled stream frame rate
                             apply_filter(color_map);   // Find and colorize the depth data

        // The show method, when applied on frameset, break it to frames and upload each frame into a gl textures
        // Each texture is displayed on different viewport according to it's stream unique id
        app.show(data);

        auto duration = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::high_resolution_clock::now() - start).count();
        std::cout << duration << " ms a cycle \n";
    }
`
**console outout :**
#### RS Frame Rate Printer ####
742 ms a cycle
27 ms a cycle
20 ms a cycle
31 ms a cycle
33 ms a cycle
32 ms a cycle
31 ms a cycle
33 ms a cycle
32 ms a cycle
32 ms a cycle
32 ms a cycle
32 ms a cycle
33 ms a cycle
31 ms a cycle

MartyG-RealSense, 2022-08-26:
I wonder whether accessing wait_for_frames() via **NativeMethods** could be slowing the C# version of the application.

The SDK's C# **Capture** examples accesses wait_for_frames(0 via a **using** instruction instead.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/tutorial/capture/Window.xaml.cs#L94

Here is the front page for the Capture example.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/tutorial/capture

jinneigeyu, 2022-08-29:
yes,  i test code from  SDK's C# Capture  example . and not found other operations  in Wrappers and C Interface。Can't understand why  NativeMethods  could be slowing

MartyG-RealSense, 2022-08-29:
NativeMethods provides a way of accessing from C# code the **RS2::** functions within the RealSense library, as described at the documentation link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/pinvoke.md

So a script that uses NativeMethods may have more processing work to do than a C++ script.

jinneigeyu, 2022-08-30:
I found the point !!!  config settings  depthProfile.Framerate is 5 ,  make it slow 
[](https://github.com/IntelRealSense/librealsense/blob/1b9a40da35f20a46b2113cabd920ce37f757743b/wrappers/csharp/tutorial/capture/Window.xaml.cs#L78)

closed 2022-08-30 by jinneigeyu