realsenseai/librealsense #10831: Intel® RealSense™ Depth Cameras D455 not working with Windows 11?

famir1, 2022-08-26:
I have installed the latest version of SDK  for D455 on Windows 11, but I cannot see any images. However, when I use Windows 10, there is no problem. 
How can I solve this problem? Is there any way to use this camera on Windows 11?

Thank you for your help and advice, in advance.

here is what i can get vis windows 11
[image]

MartyG-RealSense, 2022-08-26:
Hi @famir1  The RealSense SDK and RealSense cameras do work with Windows 11, though - like with Windows 10 - some RealSense users may experience issues on particular computers that other users might not experience on their own computers.

Can you check the camera privacy settings in Windows please to ensure that pernission to access the camera hardware and permission for applications to access the camera are both set to **On**.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

MartyG-RealSense, 2022-09-02:
Hi @famir1  Do you require further assistance with this case, please?  Thanks!

famir1, 2022-09-02:
Hi@MartyG-Realsense 
I have checked the settings of the camera on Windows 11 and I was able to see the 2 cameras, but still I was getting no frame received. Finally, I decided to work with Windows 10. 
Thanks!

closed 2022-09-13 by MartyG-RealSense