realsenseai/librealsense #10832: UNKNOWN in rs2_get_option(options:0000022A68E0EF20, option:Gain): hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."

MakerADD, 2022-08-26:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows 11|

turining on the sterio module returns an error and the hardware disconnecting.

UNKNOWN in rs2_get_option(options:0000022A68E0EF20, option:Gain):
hr returned: HResult 0x8007001f: "A device attached to the system is not
functioning."

MartyG-RealSense, 2022-08-26:
Hi @MakerADD  Does the same error occur if you download and run the standalone 2.50.0 version of the RealSense Viewer (**Intel.RealSense.SDK.exe**) from the 'Assets' file list at the link below?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

MartyG-RealSense, 2022-09-08:
Case closed due to no further comments received.

closed 2022-09-08 by MartyG-RealSense

databang-io, 2022-09-23:
Hi I'm having the same issue when i start intel realsense viewer 

 I'have tested with [v2.50.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0)  and the last version .

MartyG-RealSense, 2022-09-23:
Hi @databang-io  If you are also using Windows, please first try uninstalling the two **Depth** and **RGB** camera drivers in the Device Manager interface of Windows using instructions at the link below.

In step 4 of the instructions, if your particular computer does not offer you a tick-box with the message "Delete the driver software for this device" beside it then please skip to the next stage of the instructions.