realsenseai/librealsense #10834: Unable to install librealsense libaries

Chick92, 2022-08-27:
Following https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Unable to install any of the libraries. 

`E: Unable to locate package librealsense2-dkms`

Steps taken - 

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`

` sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`

No errors are given, until i try and install librealsense2-dkms etc.

Platform - Raspberry pi 4
OS - Ubuntu 20.04 LTS
Kernel - 5.4.0-1069-raspi

I've built it from source, but it isn't working, likely because I know how conflicting installs. I'm happy to wipe the OS and start again, but i'd much rather install the libraries using apt-get. 

Is it no longer supported? I had this working pretty well the last time I used a realsense with a Pi, but admittedly, it was about 2 years ago. 

I'll give this a go on a fresh install tomorrow - https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

MartyG-RealSense, 2022-08-28:
Hi @Chick92  The dkms packages at the distribution _linux.md instruction page are for computers with x86 / x64 processor chips such as desktop and laptop PCs, whilst Raspberry Pi has an **Arm"" processor architecture.

The simple **libuvc backend** method of installing from source code at the link below is compatible with Raspberry Pi.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Alternatively, the guide at the link below may be helpful.  You should be able to skip steps 3 and 4 if you do not need screen sharing or ROS.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

The **datasith** guide that you linked to is for use with the **Raspbian / Raspberry Pi OS** operating system rather than Ubuntu.

Chick92, 2022-08-28:
Hi Marty, 

Yeah i know it was Raspberry pi OS specific, but i just tailored the instructions for Ubuntu 20.

I was able to get everything installed and successfully bring the camera colour image and depth image up, but can't seem to get the pointcloud up. I think it's the same issue as this - https://github.com/IntelRealSense/realsense-ros/issues/1324

The first link you shared looks viable, but the second has depracted instructions. 

Issue 1324 seems to point to an issue with the Kernel, therefore i think install with the libUBC blackened would be the best bet. What do you think?

MartyG-RealSense, 2022-08-28:
I would recommend trying the libuvc backend first, since it will be quick to find out whether it works or not.

Raspberry Pi does occasionally have problems with pointcloud generation in the RealSense ROS wrapper.  If you are using ROS1 (Kinetic, Melodic, Noetic) then generating a pointcloud with the **rs_rgbd*** launch file using the instructions at https://github.com/IntelRealSense/realsense-ros/issues/2454#issuecomment-1221254065 may help.

Chick92, 2022-08-28:
Thankyou so much for taking the time to help me. I really appreciate it. 

rs_rgbd works! 

I had gotten a single point cloud frame out of rs_camera launch with the point cloud filter on before the camera became unresponsive. 

Out of interest, what is going on here? 

The launch file contains the following - 

To launch Realsense with software registeration (ROS Image Pipeline and rgbd_launch):
    $ roslaunch realsense2_camera rs_rgbd.launch
Processing enabled by ROS driver:
    # depth rectification
Processing enabled by this node:
    # rgb rectification
    # depth registeration
    # pointcloud_xyzrgb generation

To launch Realsense with hardware registeration (ROS Realsense depth alignment):
    $ roslaunch realsense2_camera rs_rgbd.launch align_depth:=true
Processing enabled by ROS driver:
    # depth rectification
    # depth registration
Processing enabled by this node:
    # rgb rectification
    # pointcloud_xyzrgb generation

I read that as the without aligned depth = true, the pi does the processing, and with, the camera does the processing? how is this different from rs_camera launch (functionally, not programmatically).
rs_rgbd_launch with and without aligned depth = true seem to operate at about the same update rate.

MartyG-RealSense, 2022-08-29:
rs_camera publishes the pointcloud at the topic **/camera/depth/color/points**

rs_rgbd publishes it at **depth_image_proc**

http://wiki.ros.org/depth_image_proc

MartyG-RealSense, 2022-09-04:
Hi @Chick92  Do you require further assistance with this case, please?  Thanks!

Chick92, 2022-09-05:
No, thankyou for your help. It's much appreciated.

closed 2022-09-05 by Chick92

MartyG-RealSense, 2022-09-05:
You are very welcome, @Chick92 - thanks very much for the update!

longlifeisland, 2023-04-06:
> The simple **libuvc backend** method of installing from source code at the link below is compatible with Raspberry Pi.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

I just wanted to add an information, that this also works for ROS2 Foxy release (I am also using the same OS and Raspberry mentioned by the Author of this post). 

Thanks to both of you! :)

Best wishes.

MartyG-RealSense, 2023-04-06:
You are very welcome.  Thanks so much for sharing your experience with the RealSense community!