realsenseai/librealsense #10836: ExternalException: rs2_set_option(options:000002443EF42050, option:Filter Magnitude, value:0)

Drumaddict182, 2022-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello, i'm trying to update the version of the SDK in my Unity project (I'm on windows 10) but I have this error : "ExternalException: rs2_set_option(options:000002443EF42050, option:Filter Magnitude, value:0)
Rethrow as ArgumentException: Given value 0 is outside [,] range!"

Someone know how to fix it ? Thank you very much.

MartyG-RealSense, 2022-08-28:
Hi @Drumaddict182  Which method are you using to update the Unity version in your project, please?

For example, are you downloading the RealSense installation package **Intel.RealSense.unitypackage** for a particular RealSense SDK version from the SDK's 'Releases' page and then launching the file with a double-leftclick whilst your Unity project is open to import the RealSense files?

https://github.com/IntelRealSense/librealsense/releases

[image]

Drumaddict182, 2022-08-29:
Hi @MartyG-RealSense, thanks for the quick reply. Yes , I have downloaded the unitypackage from the SDK releases page and import it in Unity via "assets > import package > custom package". First, i had an error windows saying that i don't have permission to erase the realsense2.dll file. So i closed Unity and erased the entire "RealSenseSDK2.0" directory and then I restarted Unity and imported the package again. No issue during the installation but at the end i have this "ExternalException: rs2_set_option(options:000002443EF42050, option:Filter Magnitude, value:0)" error.

MartyG-RealSense, 2022-08-29:
I attempted to replicate your issue by deleting the RealSenseSDK2.0 directory and reinstalling the 2.51.1 Unity package.  Like yourself, the installation was fine but I did not experience the ExternalException error.

Does your Unity project have any scripts you created yourself that are dependent on the RealSense Unity files and would have been left behind when the RealSenseSDK2.0 directory was deleted?

Does the error prevent you from running the project?

Does it occur if you start a new project and import the Unity package?

Drumaddict182, 2022-08-30:
I'm using a prefab from Keijiro Takahashi to add VFX to the pointcloud. It used to work with an older version of the SDK. I have  started a new project, imported the last version of the SDK and copied the files from my old project but i have the same error. Here is a screenshot of the processing chain from the prefab. 
[image]

MartyG-RealSense, 2022-08-30:
I know the project that you are referring to.

https://github.com/keijiro/Rsvfx

According to the notes on the file update history in the project's GitHub, it could be used with RealSense SDK version **2.20.0**.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.20.0

This SDK version would not be compatible with the D455 camera model, so if it worked for you successfully in the past then do you have a D415 or D435 / D435i camera, please?

It may not necessarily be an issue with the SDK though, as I note that you are using Unity 2021.  Some RealSense official example scene demos worked up until **Unity 2019** (which is what the Keijiro project was using) but had errors when Unity 2020 was introduced.  The SDK was later updated to be 2020 / 2021 compatible, but of course the Keijiro project was not updated.

Drumaddict182, 2022-08-31:
Yes that's the project, I'm using in my Unity. I have a D415 camera. I was able to upgrade the project of Keijiro from Unity 2019 to Unity 2021, it was a little tricky but it works. So i guess this project can work with the SDK 2.20 and not with the SDK 2.51 but it's fine for me. Thanks you.

MartyG-RealSense, 2022-08-31:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update!