realsenseai/librealsense #10837: Frames didn't arrived!

russel-autolife, 2022-08-29:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D415|
|**Serial Number**|020122061939|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|020223022453|
|**Firmware Update Id**|020223022453|

Please provide a description of the problem

MartyG-RealSense, 2022-08-29:
Hi @russel-autolife  Could you provide more information about your problem, please?  Did you experience the error in the **RealSense Viewer** tool or in a program that you wrote yourself?

I note from the information provided above that your camera is on a USB 2.1 connection.  This has slower data transfer speed than a USB 3 connection.  Does the error stop occurring if you use a lower resolution or frames per second (FPS) speed?  Thanks!

russel-autolife, 2022-08-29:
I sincerely appreciate your helpful assistance with the raised issue. To your point, I have realized that utilizing a USB-C connector worked better than my Intel NUC's USB 3 port because it has a lower data transmission rate. Confirming the issue is resolved and now can successfully stream the depth data.

MartyG-RealSense, 2022-08-29:
That's great news that you achieved a solution, @russel-autolife - thanks very much for the update!

MartyG-RealSense, 2022-09-05:
Hi @russel-autolife  Do you require further assistance with this case, please?  Thanks!

russel-autolife, 2022-09-05:
Good morning @MartyG-RealSense! I truly appreciate your invalubale support and the kind folow up. Confirming the issue is resolved and now the depth stream is up and running. 

Have a great rest of your day!

MartyG-RealSense, 2022-09-05:
You are very welcome, @russel-autolife - I'm pleased that I could help.  Thanks very much for the update!

closed 2022-09-06 by MartyG-RealSense