realsenseai/librealsense #10838: Camera Calibration SR300

ericwyy, 2022-08-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 16}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<5>.<0> }                          | 
| Language                            |  {ROS}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hello,
I am trying to use the camera to calculate some point clouds for robot recognition and capture.
I've started by checking the correctness of the camera's intrinsic parameters by running the calibration using this package.

When it is started with a ROS launch file:roslaunch realsense2_camera rs_camera.launch.

The following problems occurred:

> [ INFO] [1661762355.035901791]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1661762355.035913686]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1661762355.035928177]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
> [ INFO] [1661762355.035955195]: num_filters: 0
> [ INFO] [1661762355.035973003]: Setting Dynamic reconfig parameters.
> [ INFO] [1661762355.070243163]: Done Setting Dynamic reconfig parameters.
> [ INFO] [1661762355.070704830]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
> [ INFO] [1661762355.073319176]: color stream is enabled - width: 1920, height: 1080, fps: 30, Format: RGB8
> [ INFO] [1661762355.073466337]: setupPublishers...
> [ INFO] [1661762355.076658232]: Expected frequency for depth = 30.00000
> [ INFO] [1661762355.097935356]: Expected frequency for color = 30.00000
> [ INFO] [1661762355.119495075]: setupStreams...
> [ INFO] [1661762355.141283367]: SELECTED BASE:Depth, 0
> [ INFO] [1661762355.148400445]: RealSense Node Is Up!
> [ WARN] [1661762355.500953322]: 
> [ WARN] [1661762355.500985720]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
> [ERROR] [1661762355.787163711]: **Region-of-interest is not implemented for this device!**
>  29/08 17:02:27,629 WARNING [139814460589824] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds

Thanks！

MartyG-RealSense, 2022-08-29:
Hi @ericwyy  The SR300 camera model does not have an official calibration tool.  You can obtain the camera's intrinsics using the librealsense SDK's 'rs-enumerate-devices' tool by launching it in the Ubuntu terminal in calibration information mode with the command **rs-enumerate-devices -c**

[image]

In regard to the ROS warning, does it still occur if you reset the camera at launch by adding **initial_reset:=true** to the roslaunch instruction?

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

ericwyy, 2022-08-29:
Thank you for your reply！
I have revised it according to your reply, but there are still some problems.
When I print **rs-enumerate-devices -c** in the Ubuntu terminal
it got wrong:rs-enumerate-devices-c：未找到命令
and just **rs-enumerate-devices** in the Ubuntu terminal, something appeared as:
Device info: 
    Name                          : 	Intel RealSense SR300
    Serial Number                 : 	616205003322
    Firmware Version              : 	3.21.0.0
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.2/video4linux/video1
    Debug Op Code                 : 	53
    Product Id                    : 	0AA5
    Camera Locked                 : 	YES
    Product Line                  : 	SR300
    Asic Serial Number            : 	616205003322
    Firmware Update Id            : 	616205003322

Stream Profiles supported by Coded-Light Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  640x480	@ 60Hz	   Y16
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y16
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 10Hz	   Y16
    Infrared 1	  640x480	@ 10Hz	   Y8
    Infrared 1	  640x240	@ 110Hz	   Y16
    Infrared 1	  640x240	@ 110Hz	   Y8
    Infrared 1	  640x240	@ 60Hz	   Y16
    Infrared 1	  640x240	@ 60Hz	   Y8
    Infrared 1	  640x240	@ 30Hz	   Y16
    Infrared 1	  640x240	@ 30Hz	   Y8
    Infrared 1	  640x240	@ 10Hz	   Y16
    Infrared 1	  640x240	@ 10Hz	   Y8
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 10Hz	   Z16
    Depth	  640x240	@ 110Hz	   Z16
    Depth	  640x240	@ 60Hz	   Z16
    Depth	  640x240	@ 30Hz	   Z16
    Depth	  640x240	@ 10Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 10Hz	   RGB8
    Color	  1920x1080	@ 10Hz	   Y16
    Color	  1920x1080	@ 10Hz	   BGRA8
    Color	  1920x1080	@ 10Hz	   RGBA8
    Color	  1920x1080	@ 10Hz	   BGR8
    Color	  1920x1080	@ 10Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
    Color	  1280x720	@ 30Hz	   BGRA8
    Color	  1280x720	@ 30Hz	   RGBA8
    Color	  1280x720	@ 30Hz	   BGR8
    Color	  1280x720	@ 30Hz	   YUYV
    Color	  1280x720	@ 10Hz	   RGB8
[... 107 lines cut ...]
    Color	  320x240	@ 10Hz	   RGB8
    Color	  320x240	@ 10Hz	   Y16
    Color	  320x240	@ 10Hz	   BGRA8
    Color	  320x240	@ 10Hz	   RGBA8
    Color	  320x240	@ 10Hz	   BGR8
    Color	  320x240	@ 10Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 10Hz	   RGB8
    Color	  320x180	@ 10Hz	   Y16
    Color	  320x180	@ 10Hz	   BGRA8
    Color	  320x180	@ 10Hz	   RGBA8
    Color	  320x180	@ 10Hz	   BGR8
    Color	  320x180	@ 10Hz	   YUYV
and add **initial_reset:=true** to the roslaunch, he problem(Region-of-interest is not implemented for this device!) has not changed！

MartyG-RealSense, 2022-08-29:
Did you place the space before -c when entering **rs-enumerate-devices -c** into the Ubuntu terminal, please?

The SR300 does not have the auto-exposure Region of Interest feature, so the message about it being implemented is not something to be concerned about (unless you wanted to define an auto-exposure ROI area).

ericwyy, 2022-08-30:
> Did you place the space before -c when entering **rs-enumerate-devices -c** into the Ubuntu terminal, please?
> I apologize that I didn't place the space before -c exactly. When entering **rs-enumerate-devices -c** into the Ubuntu terminal, The device info below is taken from rs-enumerate-devices with an SR300.
Device info: 
    Name                          : 	Intel RealSense SR300
    Serial Number                 : 	616205003322
    Firmware Version              : 	3.21.0.0
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.2/video4linux/video2
    Debug Op Code                 : 	53
    Product Id                    : 	0AA5
    Camera Locked                 : 	YES
    Product Line                  : 	SR300
    Asic Serial Number            : 	616205003322
    Firmware Update Id            : 	616205003322

Stream Profiles supported by Coded-Light Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  640x480	@ 60Hz	   Y16
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y16
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 10Hz	   Y16
    Infrared 1	  640x480	@ 10Hz	   Y8
    Infrared 1	  640x240	@ 110Hz	   Y16
    Infrared 1	  640x240	@ 110Hz	   Y8
    Infrared 1	  640x240	@ 60Hz	   Y16
    Infrared 1	  640x240	@ 60Hz	   Y8
    Infrared 1	  640x240	@ 30Hz	   Y16
    Infrared 1	  640x240	@ 30Hz	   Y8
    Infrared 1	  640x240	@ 10Hz	   Y16
    Infrared 1	  640x240	@ 10Hz	   Y8
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 10Hz	   Z16
    Depth	  640x240	@ 110Hz	   Z16
    Depth	  640x240	@ 60Hz	   Z16
    Depth	  640x240	@ 30Hz	   Z16
    Depth	  640x240	@ 10Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 10Hz	   RGB8
    Color	  1920x1080	@ 10Hz	   Y16
    Color	  1920x1080	@ 10Hz	   BGRA8
    Color	  1920x1080	@ 10Hz	   RGBA8
    Color	  1920x1080	@ 10Hz	   BGR8
    Color	  1920x1080	@ 10Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
    Color	  1280x720	@ 30Hz	   BGRA8
    Color	  1280x720	@ 30Hz	   RGBA8
    Color	  1280x720	@ 30Hz	   BGR8
    Color	  1280x720	@ 30Hz	   YUYV
    Color	  1280x720	@ 10Hz	   RGB8
    Color	  1280x720	@ 10Hz	   Y16
    Color	  1280x720	@ 10Hz	   BGRA8
    Color	  1280x720	@ 10Hz	   RGBA8
[... 326 lines cut ...]
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

Extrinsic from "Infrared 1"	  To	  "Color" :
 Rotation Matrix:
   0.999997         0.00234246      -0.000769757   
  -0.00233787       0.99998          0.00592037    
   0.000783609     -0.00591855       0.999982      

 Translation Vector: 0.0256999991834164  -0.000733263499569148  0.00431601284071803  

Extrinsic from "Infrared 1"	  To	  "Infrared 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0  

I don't know why there are so many configurations and how to configure **roslaunch realsense 2_ camera rs_ camera. Launch** to execute normally.

> The SR300 does not support the auto-exposure Region of Interest feature, so the message about it being implemented is not something to be concerned about (unless you wanted to define an auto-exposure ROI area with an SR300).
>I don't wanted to define an auto-exposure ROI area with an SR300. I want to perform "hand eye calibration" between the camera and the robot arm based on eye to hand.

MartyG-RealSense, 2022-08-30:
Each resolution has its own set of intrinsic values but you only need to look at the values that relate to the particular resolution that you are using, and you can ignore the long **Stream profile supported** lists that come before the intrinsic listings.  

For example, because you are using an SR300 (which has stream defaults of 640x480 depth and 1920x1080 color), unless you are setting a custom stream configuration then the only depth and color intrinsic sets that are likely to be relevant are the ones below.  This is because the default stream profile of a particular camera model will be automatically applied if a custom definition is not provided by the RealSense user.  

So if you roslaunch an SR300 with **roslaunch realsense 2_camera rs_camera.launch** then 640x480 depth and 1920x1080 color, both at 30 FPS, should be used by the launch.

[image]

***********

Intrinsic of "Depth" / 640x480 / {Z16}
Width: 640
Height: 480
PPX: 310.209411621094
PPY: 245.262771606445
Fx: 472.950256347656
Fy: 472.950256347656
Distortion: Inverse Brown Conrady
Coeffs: 0.0981389656662941 0.203301951289177 0.00310043292120099 0.00264186388812959 -0.11238544434309
FOV (deg): 68.15 x 53.81

Intrinsic of "Color" / 1920x1080 / {YUYV/RGB8/BGR8/RGBA8/BGRA8/Y16}
Width: 1920
Height: 1080
PPX: 938.069213867188
PPY: 529.584289550781
Fx: 1386.97424316406
Fy: 1386.97448730469
Distortion: None
Coeffs: 0 0 0 0 0
FOV (deg): 69.37 x 42.54

MartyG-RealSense, 2022-09-05:
Hi @ericwyy  Do you require further assistance with this case, please?  Thanks!

ericwyy, 2022-09-05:
> Hi @ericwyy Do you require further assistance with this case, please? Thanks!

Thanks to all the help, I rebuilt and succeeded.

closed 2022-09-05 by MartyG-RealSense