realsenseai/librealsense #10841: RS2_USB_STATUS_OTHER on Mac M1

Zumbalamambo, 2022-08-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |  MacOS  | 

### Issue Description
I ran `sudo realsense-viewer` in my Mac. It throws the following error,

```
 30/08 18:30:56,247 WARNING [0x16bbd3000] (messenger-libusb.cpp:42) control_transfer returned error, index: 1795, error: Result too large, number: 34
 30/08 18:30:56,248 WARNING [0x16be03000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 132 error: Undefined error: 0, number: 0
 30/08 18:30:56,248 WARNING [0x16be03000] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 132 error: Undefined error: 0, number: 0
 30/08 18:30:56,262 WARNING [0x16bbd3000] (messenger-libusb.cpp:42) control_transfer returned error, index: 1795, error: Result too large, number: 34
 30/08 18:30:56,393 WARNING [0x16bbd3000] (messenger-libusb.cpp:42) control_transfer returned error, index: 1795, error: Result too large, number: 34
 30/08 18:31:31,609 WARNING [0x16be03000] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: Undefined error: 0, number: 0
 30/08 18:31:33,771 ERROR [0x16b9a3000] (dispatcher.cpp:34) Dispatcher [0x13f705d30] exception caught: recursive_mutex lock failed: Invalid argument
get_data_usb failed, error: RS2_USB_STATUS_OTHER
```

how do I fix it?

MartyG-RealSense, 2022-08-30:
Hi @Zumbalamambo  Which method did you use to install the RealSense SDK on your M1 Mac, please?  I would recommend installing using one of the two methods listed at https://github.com/IntelRealSense/librealsense/issues/10676#issuecomment-1185203673

Zumbalamambo, 2022-08-31:
@MartyG-RealSense I installed by building it from source which didn't work as it was throwing the same error. So i used the brew to reinstall and still the same error

Zumbalamambo, 2022-09-01:
@MartyG-RealSense Im using the official cable given by the Apple. What's worse is that the realsense-viewer sometimes crashes the complete system and the shuts it down. very dangerous

MartyG-RealSense, 2022-09-01:
The official cable supplied with the camera that I refer to is the "USB-C" one provided in the box when purchasing the camera, with a full size USB connector on one end and a micro-sized connector on the other end.  

Some RealSense Mac users use their own USB C to C cable instead of the supplied one, which can be more unstable than the official supplied RealSense cable.

Can you also confirm whether you are using MacOS Monterey with your M1, please?

MartyG-RealSense, 2022-09-07:
Hi @Zumbalamambo  Do you require further assistance with this case, please?  Thanks!

Zumbalamambo, 2022-09-07:
@MartyG-RealSense no... Its not fixed yet. I'm using the official cable provided by the Intel together with [this](https://www.apple.com/de/shop/product/MJ1M2ZM/A/usb%E2%80%91c-auf-usb-adapter) .

I'm using Monterey.

MartyG-RealSense, 2022-09-07:
It is probable that the problem is related to the Mac USB C to USB A adaptor, like in https://github.com/IntelRealSense/librealsense/issues/7864

This may be difficult though to confirm if your model of Mac does not have an available USB-A port to connect the camera directly to without the adaptor.  The RealSense user with the C-to-A adaptor in the above link found that the camera was only recognized if all other USB 3 devices were unplugged.

Zumbalamambo, 2022-09-13:
@MartyG-RealSense yes, still the problem is not fixed. I would wait for the next update.

MartyG-RealSense, 2022-09-13:
The RealSense SDK updates on an infrequent schedule, so it may be at least 6 months from the time of writing this before the next release as SDK version 2.51.1 was published recently.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

Zumbalamambo, 2022-09-15:
@MartyG-RealSense thank you!... I will wait :)

MartyG-RealSense, 2022-09-21:
Given the length of time until the next SDK update at the time of writing this, I would recommend closing this case and revisiting it at a later date.  Thanks for your patience!

Zumbalamambo, 2022-09-21:
:(

Zumbalamambo, 2022-09-21:
@MartyG-RealSense :(

MartyG-RealSense, 2022-09-21:
I am sorry.  In https://github.com/IntelRealSense/librealsense/issues/10841#issuecomment-1248014863 I thought that you meant that you did not mind waiting 6+ months until the next SDK release after 2.51.1.  I apologize if I misunderstood.

closed 2022-10-03 by MartyG-RealSense