realsenseai/librealsense #10842: Plan to Support Jetson Running on 20.04

dverstandig, 2022-08-30:
Hello,

I am running a jetson using the new jetpack version (Jetpack 5.0.2) in ubuntu 20.04 and was wondering if there are plans to support librealsense2 debian package in 20.04 arm64 architecture. 

Thanks,
Dylan

MartyG-RealSense, 2022-08-31:
Hi @dverstandig  There are plans to support Arm64 Debian packages for 20.04 as part of the RealSense SDK's next generation of networking system, though I do not have a time estimate for release of this feature at the time of writing this.

MartyG-RealSense, 2022-09-06:
The latest information that I have now from my Intel RealSense colleagues after discussing the issue with them is that Arm64 package support for Ubuntu 20.04 will not be added in the foreseeable future unfortunately and the SDK must therefore be compiled from source code for Arm64 on Ubuntu 20.

MartyG-RealSense, 2022-09-11:
Hi @dverstandig  Do you require further assistance with this case, please?  Thanks!

dverstandig, 2022-09-11:
Hi @MartyG-RealSense,

Thank you for letting me know. I’ll go ahead and close this for now.

closed 2022-09-11 by dverstandig