realsenseai/librealsense #10843: D435i Error, Hardware Notfication: mOtion Module force pause,Hardware error

goyalvarun02, 2022-08-31:
[image]
Hi @MartyG-RealSense 

I am using another setup with D435i , When I run the roslaunch command with initial_reset:=true observed Hardware error .

Please find the output of roslaunch, Rs-enumerate-device .
[image]
[image]

MartyG-RealSense, 2022-08-31:
As you do not have the IMU (motion module) topics 'gyro' and 'accel' enabled in your roslaunch, the **motion module force pause** warning will likely be harmless and can be ignored.

The repeating **control_transfer returned** warning is likely because there is a communication problem with the camera, perhaps related to the USB system (the USB port or the USB cable).

I see that you are already resetting the camera at launch with **initial_reset:=true**

Do the errors still occur if you use 640x480 resolution for both depth and color?

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true depth_width:=640 depth_height:=480 depth_fps:=30.0 color_width:=640 color_height:=480 color_fps:=30.0**

goyalvarun02, 2022-08-31:
Hi @MartyG-RealSense 
But it slow down the PC , Need to restart the terminal to fire the command. 
But still the same issue observed with your command .
[image]
[image]

MartyG-RealSense, 2022-08-31:
It looks as though the launch believes that the defined color stream in the custom launch instruction that I provided is width 640, height 640 instead of the 640, 480 in my instruction.  It is therefore declaring the custom configuration invalid and instead applying depth 640x480 and color 1280x720 (the default stream configuration of the camera).

Can you check please that the roslaunch instruction that you used contained **color_width:=640 color_height:=480** please

closed 2022-09-13 by MartyG-RealSense