realsenseai/librealsense #10845: use Csharp ,  when calling  Frame.As<T> , a  Debug.Assert occurs

jinneigeyu, 2022-08-31:
```
  var pp = pipeline.Start((frame) =>
                        {
                            if (frame.Profile.Stream == Stream.Depth)
                            {
                                using (var color = frame.As<DepthFrame>())
                                {
                                    // other ops
                                }
                            }

                        });
```

https://github.com/IntelRealSense/librealsense/blob/1b9a40da35f20a46b2113cabd920ce37f757743b/wrappers/csharp/Intel.RealSense/Frames/VideoFrame.cs#L17

'(NativeMethods.rs2_is_frame_extendable_to(ptr, Extension.VideoFrame, out error) > 0);'

MartyG-RealSense, 2022-08-31:
Hi @jinneigeyu  It looks as though you may have an additional bracket on your pipe start line (shown below) that should be removed.

[image]

jinneigeyu, 2022-09-01:
Hi **@MartyG-RealSense** ,That doesn't seem right . it a lamda as a  FrameCallback parma for  ` public PipelineProfile Start( FrameCallback cb);`

> Hi @jinneigeyu It looks as though you may have an additional bracket on your pipeline start line (shown below) that should be removed.
> 
> [image]

MartyG-RealSense, 2022-09-01:
It appears that the highlighted bracket ( that is opened in the pipeline start line is closed at the end of the script with }**)** which makes it a large section of code nested within ( and ) brackets.

The approach to pipeline start typically used by the RealSense SDK's official**C# Cookbook** page of C# examples would be to define the pipeline with a **new Pipeline()** instruction using code such as the lines below.

```
var pp = new Pipeline();
pp.Start();
FrameSet frames = pp.WaitForFrames();
Frame frame = frames.FirstOrDefault(Stream.Depth);
```

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md

closed 2022-09-09 by MartyG-RealSense