realsenseai/librealsense #10848: How to generate faces with Fast unorganized mesh methode ?

Amayess, 2022-08-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win10| 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0  | 
| Language                            |  C++    | 
| Segment			|  Robot | 

### Issue Description
Hi,

In order to generate a mesh with the intel RealSense D455, I used the PCL library and more precisely the Fast Organized Mesh method. I manage to generate a mesh in OBJ format but it only contains vertices and no faces.

Here is the code I used: 

    
```
	pcl::PointCloud<pcl::PointXYZ>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZ>);
	cloud->resize(size);

	for (int i = 0; i < size; i++)
	{
		cloud->at(i).x = Cloud[i].x;
		cloud->at(i).y = Cloud[i].y;
		cloud->at(i).z = Cloud[i].z;

	}

	std::cout << "Fast mesh started " << "\n";
	pcl::OrganizedFastMesh<pcl::PointXYZ>::Ptr orgMesh(new pcl::OrganizedFastMesh<pcl::PointXYZ());
	pcl::PolygonMesh triangles;
	orgMesh->setTriangulationType(pcl::OrganizedFastMesh<pcl::PointXYZ>::TRIANGLE_RIGHT_CUT);
	orgMesh->setInputCloud(cloud);
	orgMesh->reconstruct(triangles);
	std::cout << "start saving the fast mesh obj..." << "\n";
	pcl::io::saveOBJFile("Mesh.obj", triangles);
```

Can someone explain how to make the method generate faces too? 

Thanks for your help.

MartyG-RealSense, 2022-08-31:
Hi @Amayess  It looks as though the version of this question that you posted on StackOverflow is generating useful responses, so that may be the best place to obtain a solution.

https://stackoverflow.com/questions/73554594/how-to-generate-faces-using-the-pcl-fast-organized-mesh-method

Amayess, 2022-08-31:
hi @MartyG-RealSense,

Is the Point Cloud generated by the RealSense organized or unorganized ? 

Thank you.

MartyG-RealSense, 2022-08-31:
Whether a point cloud generated by librealsense is organized or unorganized is not usually mentioned.  In the RealSense ROS wrapper though, which runs as a layer over librealsense, it is **unorganized** by default.

closed 2022-09-13 by MartyG-RealSense