realsenseai/librealsense #10853: D435 camera cannot be opened!

2513297244, 2022-09-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC |
| SDK Version                            |  {2.50 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Using SDK C + + to open the camera and get the frame display program, which is normal on the developed computer
But change the program to a computer without network and plug the camera into the computer without network
The program failed to open the camera. Tip: This is the first time that this computer without network is plugged into the d435 camera device
What is the reason? Is it because the first time you plug in a computer, you need to install some drivers on the network?

MartyG-RealSense, 2022-09-01:
Hi, are you using the **re-server** ethernet networking system to network the camera, please?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

If you are using rs-server then it sounds as though there may be instructions in the C++ code for treating the camera as a **Network Device** (a networked camera) and so it does not understand how to operate as a normal camera outside of a network.  In such a script, commenting out the line of code that defines the camera as a Network Device - such as rs2::net_device** - should enable it to function outside of a network.

2513297244, 2022-09-01:
> 

The USB 3.0 connection is used,
When the d435 camera is connected to a new computer for the first time through USB 3.0, will it automatically download some files or drivers Online

MartyG-RealSense, 2022-09-01:
Thanks very much for the clarification.  When a camera is attached to a Windows computer for the first time, the computer will install two **RGB** and **Depth** drivers that can be viewed in the **Cameras** section of the Windows Device Manager interface.  It will only install the drivers for that particular camera if they are not currently installed.  The image below shows the example of drivers for the D455 camera model listed in the Device Manager.

[image]

I performed a test where I disconnected from the internet, uninstalled the drivers and then unplugged the camera and re-inserted it to initiate a driver installation.  The drivers reinstalled successfully without an internet connection.

2513297244, 2022-09-01:
Thank you very much for your reply. I will repeat your experiment and find the problem of my own program

2513297244, 2022-09-02:
> Thanks very much for the clarification. When a particular camera is attached to a Windows computer for the first time, the computer will install two **RGB** and **Depth** drivers that can be viewed in the **Cameras** section of the Windows Device Manager interface. It will only install the drivers for that particular camera upon insertion into the USB port if they are not currently installed. The image below shows the example of drivers for the D455 camera model listed in the Device Manager.
> 
> [image]
> 
> I performed a test where I disconnected from the internet, uninstalled the drivers and then unplugged the camera and re-inserted it to initiate a driver installation. The drivers reinstalled successfully without an internet connection.

OK，I have found the problem. Let me ask you a question: is there an excuse for the SDK to obtain whether the current camera is connected to a USB 2.0 or USB 3.0 port?

2513297244, 2022-09-02:
> OK，I have found the problem. Let me ask you a question: is there an excuse for the SDK to obtain whether the current camera is connected to a USB 2.0 or USB 3.0 port?

I have found the interface. I think this topic can be closed. Thank you for your reply!

MartyG-RealSense, 2022-09-02:
The USB port can decide whether a connection is USB 2.1 or USB 3 based upon the connector's pins, as USB 3 cables have additional wires and connector pins that USB 2.1 cables do not have.

Even if a USB 3 cable is used though, it may be detected as USB 2.1 if the connector is not inserted all of the way into the port.

Sometimes though it is mis-detected for unknown reasons, such as showing as USB 3,2 in the RealSense Viewer program but 2.1 in the RealSense ROS wrapper.

I'm pleased to hear that you found a solution.  Thanks very much for the update.  As you suggested, I will close the case.  Thanks again!

closed 2022-09-02 by MartyG-RealSense