realsenseai/librealsense #10854: How to use the calibrated intrinsic and extrinsic parameters of the depth camera and color camera to align the depth image to the color image ？

Huskywyh, 2022-09-01:
How to use the calibrated intrinsic and extrinsic parameters of the depth camera and color camera to align the depth image to the color image ？

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 / D400 } | 
| Operating System & Version |  Win10 | 
| Platform                                 | PC |
| SDK Version                            | 2.50 | 
| Language                            | C++/python/opencv/C++/unity | 

### Issue Description
I want to calibrate the camera first to get the accurate intrinsic and extrinsic parameters of the depth camera and color camera. I want to know how to align the depth image to the color image by using the calibrated parameters instead of default parameters ?

MartyG-RealSense, 2022-09-01:
Hi @Huskywyh  There are two official tools for calibrating RealSense cameras.  Both can write a calibration to the camera hardware.

1.  The **Dynamic Calibration** software package, which provides a robust calibration of both depth and RGB sensors.  A Windows installer file for the tool can be obtained at the link below.

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

This version of the tool only calibrates extrinsics.  If calibration of intrinsics is required then an **OEM** version of the tool that calibrates both intrinsics and extrinsics is provided with Intel's $1500 USD **OEM Calibration Target** system, though calibration of extrinsics only with the free version of the tool is fine for the majority of RealSense users.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

2.  The **On-Chip** calibration tool, which is built into the RealSense Viewer program and can be accessed from the **More** option near the top of the Viewer's options side-panel.

[image]

A guide document for the On-Chip tool can be found at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

The On-Chip tool can calibrate intrinsics and extrinsics, though as separate calibration operations rather than both simultaneously.

*********

Depth-color alignment is performed separately from calibration operations.  Are you planning on using C++, C# or Python for alignment please?

Huskywyh, 2022-09-01:
Thank you! I want to know if there a way to write calibration parameters to the camera by using c++ or python so that I can directly use the rs2::align class to realize the align process ?

MartyG-RealSense, 2022-09-01:
You can control calibration with C++ or Python scripting with the On-Chip tool.

**C++**
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#appendix-d-on-chip-calibration-c-api

**Python**
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#appendix-c-on-chip-calibration-python-api

With both C++ and Python, saving a new calibration permanently to the device can be done via **cal.write_calibration()** after calling **set_calibration_table**

closed 2022-09-13 by MartyG-RealSense