realsenseai/librealsense #10855: How to get all data using pyrealsense2.pyrealsense

PierreLouisT, 2022-09-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-09-01:
Hi @PierreLouisT  If you are seeking to query the **RGB** saturation option then that would be inaccessible with depth_sensor, which queries the depth sensor.  You would have to define a separate color_sensor option that queries the RGB sensor.

If you are querying sensors by their index number with a **query_sensors** instruction then '0' is the depth sensor and '1' is the RGB sensor.

If you are querying the sensor with a 'first sensor' instruction like **get_device().first_depth_sensor()** then you can create a **get_device().first_color_sensor()** for the RGB sensor, as demonstrated in a Python script at https://github.com/IntelRealSense/librealsense/issues/10002#issuecomment-981413044

**********

The **Advanced Mode** functions are not documented, as they interact with each other in complex ways and so Intel chose to control them with machine-learning algorithms instead of creating documentation.  If you would like information on one particular Advanced Mode function then I will be happy to discuss it with my Intel RealSense colleagues.

Intel's camera tuning white-paper guide at the link below is an excellent reference for learning how to adjust some of the RealSense settings.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

ASIC temperature refers to the temperature of the camera's internal circuit board during operation, whilst projector temperature refers to the temperature of the camera's infrared projector component if that particular camera model is equipped with one.  

Other temperature types such as MC, LLD and APD do not have official documented descriptions, though they are ones that are usually never needed to be used in a program, with ASIC and Projector being the most important ones.

An example of code for querying the temperatures with get_option via Python is provided at the link below.

https://support.intelrealsense.com/hc/en-us/articles/360053345793-How-to-query-device-temperatures-in-Python

PierreLouisT, 2022-09-01:
Hi Marty ! 
Thank you for your  clear response, effectively I made error, I used depth sensor instead of color sensor, so I understand better ! 
Ok I understand better why there is no documentation for advanced mode functions.
If I have any others questions, I come back here, 
See you soon

closed 2022-09-01 by PierreLouisT

MartyG-RealSense, 2022-09-01:
You are very welcome, @PierreLouisT - please do feel free to ask further questions if you have them!