realsenseai/librealsense #10856: Error

Nimma-shravan-kumar-reddy, 2022-09-01:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2022-09-01:
Hi @Nimma-shravan-kumar-reddy  Does this error occur as soon as the RealSense Viewer is launched or when making an option change such as enabling the depth stream, please?  For example, in https://github.com/IntelRealSense/librealsense/issues/10249 a RealSense user found that it would occur for them when selecting 1920x1080 as the depth resolution.

Nimma-shravan-kumar-reddy, 2022-09-01:
Now I don't see the error message.

closed 2022-09-13 by MartyG-RealSense