realsenseai/librealsense #10859: Autocompletion for pyrealsense2 not Working

yvzslmozky, 2022-09-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | RealSense Viewer v2.50.0| 
| Operating System & Version |  Linux (Ubuntu 18.04.6 LTS)   | 
| Kernel Version (Linux Only)    |  5.4.0-124-generic                                        | 
| Platform                                 | PC/ NVIDIA Jetson Nano  |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  python                          | 
| Segment			|  Robot                        | 

### Intellicode autocompletion not working for pyrealsense2
Hi,

I am using VSCode and Intellicode extension. While importing popular libraries like cv2 and numpy, not facing any problem. Also, pyrealsense2 is working well if one knows how to use python wrapper of librealsense and its functions since I have already set PYTHONPATH. However, I cannot see all the functions and data types of pyrealsense2 due to nonfunctionality of autocompletion. What can I do?

Selim

[image]
[image]

@d435i @d455 @Linux @python

MartyG-RealSense, 2022-09-02:
Hi @yvzslmozky  This issue with auto-complete not working for pyrealsense2 in VSCode has previously been discussed at https://github.com/IntelRealSense/librealsense/issues/10231

No solution was found, though one RealSense user in that discussion reported at https://github.com/IntelRealSense/librealsense/issues/10231#issuecomment-1040329452 that in their particular case autocomplete worked with some SDK instructions but not others.  Like yourself, it was reported that the issue did not occur with numpy and OpenCV.

closed 2022-09-05 by yvzslmozky

xpy-zju, 2023-03-20:
I still struggle with this issue, but this may help.
https://zhuanlan.zhihu.com/p/549452162
I found when Pylance is off, the autocompletion of pyrealsense2 works well, but numpy, cv2 will not be autocompleted.
Is there anyone who can explain this and find some solution?