realsenseai/librealsense #10862: D435i camera has many errors when I start the launch file

001zwzwzw, 2022-09-04:
hello .when I run rhe command roslaunch realsense2_camera rs_camera.launch,
camera firmware version is 5.12.9,librealsense version is  2.40.0,realsense-ros version is 2.2.20,I need your help,thanks!
the camera can start normally,but a few moment later,the terminal will send out the error message as follow,my camera has some problems?: 
[image]
[image]
<img width="592" alt="1662229663280" src="https://user-images.githubusercontent.com/55043568/188294604
[image]

[image]
-4c08b19d-41a0-4517-90a2-0a21f9520914.png">

001zwzwzw, 2022-09-04:
@MartyG-RealSense can you help me to solve the problem?Thanks!

MartyG-RealSense, 2022-09-04:
Hello, which method did you use to install the librealsense SDK and the RealSense ROS wrapper?  Did you install them both from packages at the same time with the ROS wrapper's **Method 1** instructions or build them separately (installing librealsense first and the wrapper second).

A conflict can occur if librealsense is built first and then the ROS wrapper Method 1 instructions are used, because Method 1 installs librealsense in addition to the wrapper and so two different conflicting installations of librealsense can consequently be installed on the same computer.

The log message **likely requires patch for fourcc code RW16!** suggests that librealsense may have been built from source code on Ubuntu but the kernel did not have a patch applied before the SDK was built.  If the built librealsense from source code, did you patch the kernel first using step 4 of the **Prepare Linux Backend and the Dev. Environment** section of the instructions at the link below, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

001zwzwzw, 2022-09-04:
I built realsense SDK and realsense-ros with source code.  i built them separately (installing librealsense first and the wrapper second.Do you means than the realsense SDK has conflict with realsense-ros?
If i do not have patch,i need to uninstall realsense SDK and build them again?

MartyG-RealSense, 2022-09-04:
It sounds as though your librealsense and ROS wrapper builds were built correctly and in the correct order, and should not be in conflict.

If the kernel is not patched then librealsense would not be able to perform certain functions, such as providing hardware timestamps.

You should be able to patch your kernel at any time without having to reinstall everything again.

001zwzwzw, 2022-09-05:
Hello,i have run the command as step 4,in  the final info the message:fatal:您需要指定一个标签名称.
as the message,i success? @MartyG-RealSense

MartyG-RealSense, 2022-09-05:
I believe that the fatal message that you are receiving is **You need to specify a tag name**

Which computer / computing device are you using please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).  This error message has occurred in the past at https://github.com/IntelRealSense/realsense-ros/issues/1110 when building librealsense from source code on Nvidia Jetson, as Jetsons have a different patch script when building the SDK from source.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

001zwzwzw, 2022-09-05:
i see the issue IntelRealSense/realsense-ros/issues/1110,but i can not get a corrent method!

MartyG-RealSense, 2022-09-05:
If you are using Ubuntu then the link below has a guide for installing librealsense on a Pi 4 with ROS Noetic and Ubuntu 20.04.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

The guide does not include instructions for installing the RealSense ROS wrapper, but as far as I am aware you should be able to install the wrapper from source code on Pi 4 using the **Method 2** instructions.

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

***********

An alternative method of installing librealsense from source code on Ubuntu that is Pi compatible is the simple **libuvc backend** installation procedure here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

001zwzwzw, 2022-09-05:
my ubuntu version is 18.04,youe method can be used in ubuntu-18.04?

MartyG-RealSense, 2022-09-05:
The libuvc backend method should be able to be used with 18.04.

001zwzwzw, 2022-09-05:
should I uninstall my previous librealsense SDK?

MartyG-RealSense, 2022-09-05:
Yes, I would highly recommend uninstalling the previous SDK.

If you built it from source code with CMake, you can do so by going to the **build** folder of your librealsense installation and running the command below:

**sudo make uninstall && make clean**

001zwzwzw, 2022-09-05:
[image]
i run the sh file,but now i can not launch the camera,it tell me that i do not have .so file as the picture.i face with new problem

MartyG-RealSense, 2022-09-05:
After running the .sh file, did you rebuild the ROS wrapper?  Every time that librealsense is installed, the ROS wrapper must be built again afterwards.

closed 2022-09-17 by MartyG-RealSense