realsenseai/librealsense #10863: The Relationship Between Time And Camera data

FeiSong123, 2022-09-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 5.4.0-124-generic                | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                          | 
| Language                            | C++                       | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

My L515 is on a robot and it keeps moving. I'm a little confused about the relationship between time and camera data. Let's look at the following code:

frameSet = pipe.wait_for_frames();											//等待相机数据	
auto process = alignToColor.process(frameSet);								//对齐处理

rs2::video_frame colorFrame = process.get_color_frame();					        //获取彩色帧
rs2::depth_frame alignedDepthFrame = process.get_depth_frame();				//获取深度帧

When I get colorFrame  and alignedDepthFrame, is the time corresponding to the camera data before wait_for_frames or after get_depth_frame ?

MartyG-RealSense, 2022-09-04:
Hi @FeiSong123  A RealSense team member has offered the advice below in the past for similar questions to yours.

************

The timestamps of the acquired frames will be **older** than the time that the wait_for_frames call was made.  The frame's timestamp refers to either (a) the time point that the frame was generated in Hardware or (b) the time it was acquired by the RealSense SDK.

The frames are processed internally in asynchronous mode.  The call wait_for_frames does not activate the processing pipeline in Host/Device but is merely a consumer asking for the latest frame(set) generated by the library.

Documentation about timestamp types can be found at the link below.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

***********

So as the timestamps are older than wait_for_frames, they will have come **before** it.

FeiSong123, 2022-09-04:
Can I understand that the camera data at this time (after I  get_depth_frame) corresponds in time and space is before wait_for_frames ?

MartyG-RealSense, 2022-09-04:
wait_for_frames does not return instantly but instead blocks until a complete frame is available.  So the camera data will be older than when wait_for_frames is called.

If poll_for_frames is used instead of wait_for_frames then frames are returned instantly, as described at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

FeiSong123, 2022-09-05:
Thank you for your guidance. I'll check it for more information.

MartyG-RealSense, 2022-09-11:
Hi @FeiSong123  Do you require further assistance with this case, please?  Thanks!

FeiSong123, 2022-09-11:
No,thanks !

closed 2022-09-11 by FeiSong123