realsenseai/librealsense #10864: Oculus Quest 2 with Intel L515 to visualise nearby obstacles

AndreDesira, 2022-09-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 11 | 
| Platform                                 | PC & Oculus Quest 2  |
| SDK Version                            |  2.50.0                           | 
| Language                            |  C#/unity                          | 
| Segment			|  VR                          | 

### Issue Description
Hi There. I am doing a project consisting of obstacle avoidance when using virtual reality. Since current solutions such as Chaperone and Guardian aren't that good and users still get injured, bump into things or require to move furniture aside each time they want to play, my project is to use a 3D depth sensor to avoid all that. Through the use of the Intel L515 mounted onto the VR headset I am to scan dynamically and in real time the environment, capturing 3D Point Cloud Data and then through Scripting on the PC and Unity I would filter out the point cloud data as I please, create a wireframe out of the downsampled and cleaned point cloud and the render it onto the VR headset through unity whenever the user is getting to close to the corresponding obstacle being sensed. 

The problem is that I have zero experience with unity, scripting and the use of sensors. While I was able to get the 3d point cloud data to be rendered onto unity through the SDK, I can't understand the scripting code and where I would want to insert my code. I was hopping someone could assist me into understanding a bit the scripts of the Unity SDK since the scripts themselves have little to no in line documentation and I also could not find any detailed documentation relating to Unity SDK.

I would really appreciate any help as since I am quite a beginner the whole Realsense viewer package and SDK are a bit out of my league. If I could understand a bit how the SDK works altogether and how the process of gathering the point cloud to processing it and then being rendered, I would then be able to make sense of it and deduct how and where I am to insert my code.

Thank you all in advance.

@MartyG-RealSense @matkatz

MartyG-RealSense, 2022-09-04:
Hi @AndreDesira  A good place to start your research would be a RealSense project for the original Oculus Quest that a RealSense user developed for the D435i camera model.

https://github.com/GeorgeAdamon/quest-realsense

The 'making of' discussion notes that led to this project are at https://github.com/IntelRealSense/librealsense/issues/4155

***********

When you say 'Unity SDK', do you mean that you wish to learn the basics of Unity programming in general, or learn how the scripts of the **RealSense Unity wrapper** in particular (such as RsDevice) work, please?

AndreDesira, 2022-09-04:
Hi Marty thanks for the speedy reply. I had previously looked at GeorgeAdamon's project and the #4155 . I didn't go into much detail for that issue as I can't understand most of it since I am having issues at an earlier state.

Yes, my issue is I can't make sense of the existing scripting code due to lack of said unity programming knowledge and lack of available documentation with regards to Intel Real sense Unity SDK. Or at least I wasn't able to find such documentation should it exist.

Currently what I am able to do is show in real time the coloured 3D point cloud onto the virtual environment. Refer to the first picture.

[image]

However, as can be seen in the picture, the 3D point cloud is too dense making the background difficult to see and so reducing the immersive experience. Additionally, the background wall (data in red) is also being picked up which is unnecessary as only the closest obstacle would need to be rendered onto the Oculus and only if the user is within a certain threshold, say 30cm, otherwise the 3D point cloud should not be visible as to not ruin the VR experience. Even the colours I would want to change. So circling back to the original issue is to understand how the script works so that I would know how and where to add my own code to manipulate the 3D point cloud. Then I would presume that through constantly updating and looping through the same script, whenever the user moves the same script to translate the dense 3d point cloud into a clean less dense wireframe would be run at a split of a second to constantly show the correct wireframe and data based on the user's location. I hope I managed to explain myself. 

The final result would look similar to the image below where the colour of the wireframe changes depending on the user's distance from the identified obstacle:

[image]

MartyG-RealSense, 2022-09-04:
The L515 camera model has a minimum depth sensing distance of around 25 cm.  Objects / surfaces nearer to the camera than that may not be rendered on the depth image.  So the depth range that you would be aiming to render is between 25 and 30 cm.

https://github.com/IntelRealSense/librealsense/issues/1898 has advice about how to implement **post-processing filters** in Unity, which should allow you to use a **threshold filter** to set a minimum and maximum depth sensing distance.  This would require understanding of Unity scripting to implement though.

If you have the option to change camera model then it may be easier to use a **RealSense D405**, which is designed for high accuracy and high quality depth imaging at very close range (its ideal depth range is 7 cm to 50 cm).  The D405 also has the advantage of being smaller than the L515, making it potentially easier to mount to an Oculus Quest 2 headset.  The D405 model is compatible with the RealSense Unity wrapper.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

If you need to learn the basics of Unity development from the beginning, Unity's own **Unity Learn** platform may be the best place to begin.

https://learn.unity.com/?variant=b

closed 2022-09-17 by MartyG-RealSense