realsenseai/librealsense #10868: how does the temporal filter work in the cpp api?

omerbrandis, 2022-09-05:
hello, 

I've been having problems using rs2::temporal_filter ( rs version 2.29.8  on ubuntu 16)
my code works for Hole Filling filter , this is very apparent on the result.
but running the "same" code for the Temporal filter doesn't. ( simply using the temportal Filter's process method as indicated in the documentation ).

this raises the question - 
how can the temporal filter perform its task, which requires data from several depth frames, if the process method receives only one depth frame?

thanks,
Omer.

omerbrandis, 2022-09-05:
+ i've set the persistence index to various values from 0 to 8 without any effect, while the re-viewer is showing significant changes for the same scene ).

omerbrandis, 2022-09-05:
Hello MartyG,

Thank you very much for trying to help.

previously, my set_option statements were before the pipeline start, 
i've now moved them to between the start and the wait_for_frames(), 
but the result is the same ( no effect on the depth data ).

MartyG-RealSense, 2022-09-05:
Usually after defining set_option, the filter has to be applied with a **process** instruction like the one below:

[image]

Have you applied the filter like this, please?

omerbrandis, 2022-09-05:
I have marty. :-)
Omer.

MartyG-RealSense, 2022-09-05:
Is the temporal filter applied **before** the hole-filling filter is applied, as recommended by Intel's documentation at the link below regarding the order in which to apply the different types of post-processing filter?

https://dev.intelrealsense.com/docs/post-processing-filters#using-filters-in-application-code

Does the temporal filter work if you comment out the process line for the hole-filling filter?

omerbrandis, 2022-09-06:
hello MartyG,

In the current version of the code, i do not perform any of the other pp filters.

**can we please focus on the question I asked and not debugging my code ?**
( there are several differences between my code and the example code, 
I've made my code as similar as possible to the example code.
I realize that any changes in my code & installation may be the cause for the problem.
i have tried building rs-post-processing.cpp with the intent of then commenting out the execution of the other filters, in order to make sure the example code works, but this has turned out to be more complicated  & time consuming then i though and i have not yet manged to do it.)

so i ask again,
are "we" certain that the way to use the temporal filter is using its process method on one depth frame?
can anyone confirm that this is what is done by the rs-viewer ? ( i was not able to follow the code )
can anyone explain how the temporal filter calculates the result, which is based on a frame set, while receiving only frame at a time ? ( i've noticed that temporal_filter's parent class filter has a frame_queue member, but i was not able to follow the code , i've also tried executing it on 32 frames ( of roughly the same scene ) but this to did not generate the results seen in the viewer)

:-)
Omer.

MartyG-RealSense, 2022-09-06:
If only one frame is being generated then the post-processing documentation about the temporal filter at the link below would suggest that the filter will have no historical data from previous frames to work with, because "the temporal filter is intended to improve the depth data persistency by manipulating per-pixel values based on previous frames".

https://dev.intelrealsense.com/docs/post-processing-filters#temporal-filter

The **Spatial** filter also provides a hole-filling function, so it may be worth using that instead of the temporal filter.

https://dev.intelrealsense.com/docs/post-processing-filters#spatial-edge-preserving-filter

In regard to how the RealSense Viewer applies the filters: whilst there is not information available about this, I believe that the functions in the C++-based Viewer tend to follow the methods of the individual C++ example programs, such as **rs-post-processing**

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp

omerbrandis, 2022-09-06:
I've built rs v2.29.0 from source.
changed rs-postprocessing.cpp :
commented out 
filters.emplace_back("Decimate", dec_filter);
filters.emplace_back("Threshold", thr_filter);
filters.emplace_back(disparity_filter_name, depth_to_disparity);
filters.emplace_back("Spatial", spat_filter);
added temp_filter.set_option( RS2_OPTION_HOLES_FILL ,8);
the resulting image is clearly modified by the temporal filter. ( before i added the setting of the persistency mode/index to 8, it was not apparent ).

so we are now left with the how.
:-)

MartyG-RealSense, 2022-09-06:
It's great to hear that you made significant progress!

Point '4. Temporal filtering & persistence' of the **Simple Post-Processing** section of Intel's white-paper document about post-processing provides an explanation for the workings of the temporal filter.

https://dev.intelrealsense.com/docs/depth-post-processing#simple-post-processing

The entire section is quoted below.

************

Whenever possible, it is recommend to use some amount of time averaging to improve the depth, making sure not to let “holes” (depth=0) influence the calculations.  In the 'librealsense' implementation we stick with the same type of exponential moving average (EMA) filter that we used for spatial filtering.  Like before, there will be an alpha parameter which will represent the extent of the temporal history that should be averaged. 

The benefit of this approach is that “fractional” frames can essentially be averaged.  By setting alpha=1, there will be zero filtering, but reducing alpha to near 0, will increase averaging and smoothly.  So it is not a simple discrete “average 2 or 3 or 4 frames”, but allows for much more fine grained smoothing.  Moreover, just like with the spatial averaging, it is important to add a threshold parameter, delta.  This way we try to reduce temporal smoothing near edges, and we also do not include holes (z=0) in the averaging.  Some good default values for the temporal filter are alpha=0.5 and delta=20.

Another temporal filter we offer as on option, is the “persistence filter”.  This is essentially a filter that can be used to see farther. For every depth pixel, the filter will look at the history of holes, and will retain the last valid value seen, within a specified number of frames, from 0-7.  A persistence value of 8 means that we will retain the last valid value indefinitely, while a value of 1 means only look one frame back in history if you see a hole, otherwise use the latest value (i.e. the z=0). 

So when pointing the camera in the dark into a large room, at a certain range the valid depth values will start to give way to holes.  If you use the persistence feature, you will see much farther because it will persist pixels.  In other words, this is a type of hole-filling algorithm, but it bases its guess on having seen a good value there in the past, as opposed filling based on neighboring pixels.  

Of course, all hole-filling algorithms are prone to some errors and artifacts and should be used after careful consideration of the tradeoffs.  In the specific case of this persistence filter, there will be an artifact where if you wave an object in front of the camera, you will see a different in-filling behavior depending on whether you move the object left or right on the scene.  A movement right will persist values from the object into the occlusion region, making the object wider, whereas a movement left will persist values from the background (which is better).  As with all algorithms this can be fixed with more levels of logic, but is beyond current scope and implementation.

MartyG-RealSense, 2022-09-07:
I tested the SDK C++ example program **rs-post-processing** which applies a range of filters including the temporal filter by adjusting sliders. When disabling all filters except temporal and adjusting the filter smooth alpha setting from its default of '0.4', there was no change to the image no matter which value was set by the slider. (minimum '0' or maximum '1')

I found though that if the Decimation filter was also enabled and left at its default of '2' then the temporal filter reacted correctly when its filter smooth alpha was adjusted.

The temporal filter also reacted to changes if any other combination of temporal and another filter were enabled (temporal and spatial, temporal and disparity, etc).

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp

omerbrandis, 2022-09-09:
Hello MartyG,

Thanks again for all your time & effort.

I have reviewed file temporal-filter.cpp before opening this issue, 
i'd love to discuss it :
the temporal_filter::process_frame method is simply not what i'm calling directly from my program.
I call the temporal_filter::process method, which is not implemented in this file.
but lets assume for the moment that through some other code, the parent of temporal_filter that does implement "process" calls process_frame ( the which and how of this is part of my original question )

you mentioned that temporal_filter::recalc_persistence_map stores its result in 
 // Store results
        _persistence_map = credible_threshold;
but as far as i can tell ,  temporal_filter::process_frame does not call  temporal_filter::recalc_persistence_map both directly and indirectly .

:-)
Omer.

MartyG-RealSense, 2022-09-10:
I do no have further advice to provide about the operating principles of the temporal filter beyond the discussion points and official documentation references that have been provided above, unfortunately.  I do apologize.

The Python version of the official API documentation provides the following description: 

*********

Temporal filter smooths the image by calculating multiple frames with alpha and delta settings.  Alpha defines the weight of current frame, and delta defines the threshold for edge classification and preserving.

MartyG-RealSense, 2022-09-16:
Hi @omerbrandis  Do you require further assistance with this case, please?  Thanks!

omerbrandis, 2022-09-16:
yes i would like my questions answered .
:-)

MartyG-RealSense, 2022-09-16:
There is not any further information about the temporal filter that can be supplied, regretfully, as all of the documentation for it has already been provided during this discussion.

MartyG-RealSense, 2022-09-16:
Your questions were the ones below, I believe.

********

are "we" certain that the way to use the temporal filter is using its process method on one depth frame?

can anyone confirm that this is what is done by the rs-viewer ? ( i was not able to follow the code )

can anyone explain how the temporal filter calculates the result, which is based on a frame set, while receiving only frame at a time ? ( i've noticed that temporal_filter's parent class filter has a frame_queue member, but i was not able to follow the code , 
i've also tried executing it on 32 frames ( of roughly the same scene ) but this to did not generate the results seen in the viewer)

how/where in the rs2::temporal_filter's code for the process method, does it store the data provided to it ( for use later on / how does it perform its processing)?

MartyG-RealSense, 2022-09-28:
Could you provide your thoughts please on my recommendation above to not use the temporal filter for your project?

Because the temporal filter relies on a history of frames rather than just one frame, I unfortunately cannot foresee a way to make it work in your project with a single frame.

omerbrandis, 2022-09-28:
Hello MartyG,

I prefar to stay on point.
I specifically asked for clearification about the code and nothing else in this issue/topic.

Thanks,
Omer.

MartyG-RealSense, 2022-09-28:
You asked earlier in this discussion, "how/where in the rs2::temporal_filter's code for the process method, does it store the data provided to it ( for use later on / how does it perform its processing)?"

The question of how the temporal filter performs its processing has been answered with various references.  This leaves the remaining question of how / where does the filter store the data provided to it for use later on.

Looking at the line of temporal-filter.cpp linked to below, it is probably comparing the data of the last frame **_last_frame_data()** to the stored history **_history_data()**.  

There is no other reference in the script to _history_data() though.

MartyG-RealSense, 2022-10-04:
Hi @omerbrandis  Was the advice in the comment above helpful to you, please?  Thanks!

closed 2022-10-16 by MartyG-RealSense