realsenseai/librealsense #10870: The Temporal Filter Does Not Work Well

Shannon-Law, 2022-09-06:
In my program, when I use the temporal filter to process the depth maps transferred from the RealSense D435 camera, it does not work. However, in RealSense Viewer, it works well. Why is that and what should I do to make it normal? What's more, my codes are similar to the ones listed in https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/examples/post-processing/rs-post-processing.cpp.

MartyG-RealSense, 2022-09-06:
Hi @Shannon-Law  I recommend first reading the recent case at https://github.com/IntelRealSense/librealsense/issues/10868 to see whether the problem that the RealSense user in that case is the same as the one that you are experiencing?

Are you using C++ or Python language, please?

Shannon-Law, 2022-09-07:
Okay, @MartyG-RealSense , I'm using C++.

MartyG-RealSense, 2022-09-07:
Are you using the temporal filter for the purpose of performing hole-filling, please?  If so then the **Spatial** filter could provide the same function.

https://dev.intelrealsense.com/docs/post-processing-filters#spatial-edge-preserving-filter

Shannon-Law, 2022-09-07:
No. The purpose that I use the temporal filter is to improve the stability of the depth maps and the generated point clouds. In Realsense Viewer, they are stable due to the temporal filter. But, they oscillate in my C++ program because of the fact that the filter does not work.

MartyG-RealSense, 2022-09-07:
I tested the SDK C++ example program **rs-post-processing** which applies a range of filters including the temporal filter by adjusting sliders.  When disabling all filters except temporal and adjusting the **filter smooth alpha** setting from its default of '0.4', there was no change to the image no matter which value was set by the slider. (minimum '0' or maximum '1')  

I found though that if the **Decimation** filter was also enabled and left at its default of '2' then the temporal filter reacted correctly when its filter smooth alpha was adjusted.  

The temporal filter also reacted to changes if any other combination of temporal and another filter were enabled (temporal and spatial, temporal and disparity, etc). 

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp

MartyG-RealSense, 2022-09-13:
Hi @Shannon-Law  Do you require further assistance with this case, please?  Thanks!

Shannon-Law, 2022-09-14:
No, thanks a lot!

MartyG-RealSense, 2022-09-14:
You are very welcome, @Shannon-Law - thanks very much for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-09-14 by MartyG-RealSense