realsenseai/librealsense #10871: How to cut frame?

ddov123, 2022-09-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | | 
| Operating System & Version |Win 10| 
| Kernel Version (Linux Only)    | | 
| Platform                                 | PC|
| SDK Version                            |2.50.0.0| 
| Language                            |C#                          | 
| Segment			| Robot   | 

### Issue Description

Hello!

how to cut depthframe?

I want to remove that side frame and make pointcloud.

[image]

[image]

MartyG-RealSense, 2022-09-06:
Hi @ddov123  The wide range of different colors at approximately the same distance from the camera in the side sections makes me think that they may be inaccurate depth values, perhaps because those areas of the observed scene are difficult for the camera to read depth detail from.  Setting the camera configuration preset **High Accuracy** may therefore help to filter them out from the image.

If you are using the RealSense Viewer tool then High Accuracy can be selected from the drop-down menu beside **Preset** near the top of the Viewer's options side-panel.

[image]

ddov123, 2022-09-06:
@MartyG-RealSense I mean.. crop like this
[image]

MartyG-RealSense, 2022-09-06:
Yes, I meant that using High Accuracy may be one method of cropping out the unwanted data.  The RealSense Viewer does not have a function for XY cropping, only for Z-depth, so it would be difficult to remove data from the edges of the image.

If you can program your own application then it is possible to define an XY **bounding box** and exclude any depth data outside of the box.  There is not an example of C# code available for doing so though, unfortunately.  A C++ method is described at https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157

closed 2022-09-14 by ddov123